sheaivey/rx5808-pro-diversity #27: Add pulldown resistor to RSSI.

sheaivey, 2015-09-15:
In the event a receiver looses power or RSSI gets disconnected this will ensure there are no floating reads and will make sure the other receiver fails over.

nathantsoi, 2015-09-29:
Building one of these now, any suggestions on how strong to make the pulldown. Seems like 10k might be appropriate?

sheaivey, 2015-09-29:
I'm using 100k. A 10k allowed to much current to flow to ground and the atmega received a very weak RSSI voltage. 

100k does not mess with the overall RSSI signal very much but does give a true low when RSSI is disconnected.

Good luck with your build!

nathantsoi, 2015-09-29:
Awesome, 100k it is. Nice work w/ this project. I'm stoked to get mine running.

sheaivey, 2015-10-05:
I have updated all the images to include the RSSI pulldown resistor.

closed 2015-10-05 by sheaivey