sheaivey/rx5808-pro-diversity #29: Make it "tri", "quad" or more diversity ?

yoannwyffels, 2015-09-28:
Why just 2 rx5808 ? Would it be possible to have, 3, 4, or more connected in order to use patch or directionnal antennas ?

Hasi123, 2015-09-28:
Even antenna tracking could be implemented. Two directional and one Omni antenna.

sheaivey, 2015-09-28:
How the code is currently written it can only support two receivers at them moment. However the ATMEGA328 chip has two more free analog pins. So in theory it could support a 4 receiver diversity setup. 

If the need for this is there, the best plan of action would be to make an a few arrays that handle all the rssi concerns such as:
```
// define the number if receivers
define #RECEIVER_COUNT 4
rssi_pins[RECEIVER_COUNT] = {A6,A7,A2,A3}; // what analog pin each receiver sits on
rssi_max[RECEIVER_COUNT] = {0,0,0,0}; // for calibration of each receiver
rssi_min[RECEIVER_COUNT] = {255,255,255,255}; // for calibration of each receiver
//... and so on
```
Now the diversity concerns of the code would be based on the RECEIVER_COUNT

I'll leave this open incase someone else wants to take this on.

yoannwyffels, 2015-09-29:
Thanks Sheaivey :)
Some peoples from "Airgonay" - a famous French FPV club - successfully design a triversity (and even a quadversity), you can take a look at (in french sorry): http://www.airgonay.com/forums/showthread.php?4042-vol-FPV-racing-en-5-8G-combien-de-pilots/page34

and some nice pictures of their first prototype (with 4 directionnal antennas !):
http://www.airgonay.com/forums/showthread.php?4042-vol-FPV-racing-en-5-8G-combien-de-pilots/page17

But they use a PI5V331 IC to switch video (seems very fast according to their tests, they push the switch to 50hz without any video flickers). Does the 4066 digital switch you're using will accept more than 2 video sources ?

sheaivey, 2015-09-29:
@yoannwyffels Thanks for the info on their diversity project. That PI5V331 IC is pretty impressive. However it is about $1.25 more expensive then the 4066 IC.
in comparison the PI5V331 seems to have a faster switching (10ns vs 20ns) action but out side of that they function pretty similar.

The 4066 IC can also control 4 inputs just like the PI5V331. As far as switching is concerned flickering can come from many things...
1. For example one receiver may not output the same brightness as its siblings from the same manufacture. 
2. Another cause of flicker is how fast the receiving monitor can adjust to the ~20ns gap in signal.
Those issues will cause a flicker regardless of how good the video switching is. 

Because of that the 4066D chips are more then adequate for the task and at a substantially cheaper cost.

Another reason to choose the 4066D is that it can switch any signal such as audio and is not limited to only switching video sources. How I set up my diversity circuits video and audio are switched. 

You can also find the larger version of this digital switch as well. I think they come in 2,4,8 versions.

yoannwyffels, 2015-09-30:
PI5V331 also have a lower internal resistance. I'm not sure about consequences, and even if 4066 IC is working great, a "PI5V331" version will be better :)
Regarding audio, where do you get it ? I can't find any audio "out" in RX5808 diagram ? Maybe A6.5M out ?

sheaivey, 2015-09-30:
Yeah, give it a try with the PI5V331! I may try it myself with future builds.

You are correct about A6.5. That should be all you need to connect for audio out.

closed 2016-03-02 by yoannwyffels