sheaivey/rx5808-pro-diversity #38: Chip programer for boot loader

chaz762, 2015-10-08:
I just finished building mine (ordered your boards) and I programed a adruino mini the used my hot air station to desolder it and resolder onto new board and it worked like a champ.
if someone is trying to be cheap like me and has stuff laying around.
and Great work!!!!!!!!!!!!!!!! this thing is awesome. just now gonna stuff it into the banggood snowflakes monitor (two imputs #1 for GUI and #2 is no bluescreen input)
and You sell theese on ebay for lazy people who don't like to build stuff;-)

sheaivey, 2015-10-08:
That is great! I'm really glad you got it working!

You are right about needing SPI breakout to the ATMEGA328 chip to install the boot loader.

Next rev of the board I will add SPI programmer pins instead of the FTDI pins.

sheaivey, 2015-10-27:
I have update the advanced custom board to include SPI programmer pins along with a few other updates.

closed 2015-10-27 by sheaivey