sheaivey/rx5808-pro-diversity #59: Compile the project fail

xfce, 2016-01-12:
Hi Sir
    I have download the source code and add three libraries((Adafruit_GFX, Adafruit_SSD1306,libraries(rx5808-pro-diversity-master\src\libraries)) to the Arduino libraries folder,
When I compile the project, give below errors.

oled_128x64_adafruit_screens.cpp:47:6: error: #error ("Screen size incorrect, please fix Adafruit_SSD1306.h!");
oled_128x64_adafruit_screens.cpp:44: error: 'Adafruit_SSD1306' does not name a type
oled_128x64_adafruit_screens.cpp: In member function 'char screens::begin(const char*)':
oled_128x64_adafruit_screens.cpp:58: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:58: error: 'SSD1306_SWITCHCAPVCC' was not declared in this scope
oled_128x64_adafruit_screens.cpp:65: error: 'delay' was not declared in this scope
oled_128x64_adafruit_screens.cpp:70: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:71: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:88: error: 'A7' was not declared in this scope
oled_128x64_adafruit_screens.cpp:88: error: 'analogRead' was not declared in this scope
oled_128x64_adafruit_screens.cpp:95: error: 'strlen' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::reset()':
oled_128x64_adafruit_screens.cpp:104: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:107: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::flip()':
oled_128x64_adafruit_screens.cpp:111: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::drawTitleBox(const char*)':
oled_128x64_adafruit_screens.cpp:115: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:115: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:119: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:121: error: 'strlen' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::mainMenu(uint8_t)':
oled_128x64_adafruit_screens.cpp:130: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:130: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:132: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:143: error: 'A7' was not declared in this scope
oled_128x64_adafruit_screens.cpp:143: error: 'analogRead' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::seekMode(uint8_t)':
oled_128x64_adafruit_screens.cpp:168: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:168: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateSeekMode(uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, uint8_t, bool)':
oled_128x64_adafruit_screens.cpp:201: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:201: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:201: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:239: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:239: error: 'map' was not declared in this scope
oled_128x64_adafruit_screens.cpp:241: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:242: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::bandScanMode(uint8_t)':
oled_128x64_adafruit_screens.cpp:286: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:292: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:295: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:295: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateBandScanMode(bool, uint8_t, uint8_t, uint8_t, uint16_t, uint16_t, uint16_t)':
oled_128x64_adafruit_screens.cpp:310: error: 'map' was not declared in this scope
oled_128x64_adafruit_screens.cpp:311: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:314: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:315: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:324: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:324: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:326: error: 'HEX' was not declared in this scope
oled_128x64_adafruit_screens.cpp:340: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:340: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:343: error: 'DEC' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::screenSaver(uint8_t, uint8_t, uint16_t, const char*)':
oled_128x64_adafruit_screens.cpp:360: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:361: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:363: error: 'HEX' was not declared in this scope
oled_128x64_adafruit_screens.cpp:373: error: 'A7' was not declared in this scope
oled_128x64_adafruit_screens.cpp:373: error: 'analogRead' was not declared in this scope
oled_128x64_adafruit_screens.cpp:386: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateScreenSaver(char, uint8_t, uint8_t, uint8_t)':
oled_128x64_adafruit_screens.cpp:404: error: 'A7' was not declared in this scope
oled_128x64_adafruit_screens.cpp:404: error: 'analogRead' was not declared in this scope
oled_128x64_adafruit_screens.cpp:407: error: 'map' was not declared in this scope
oled_128x64_adafruit_screens.cpp:408: error: 'display' was not declared in this scope
[... 22 lines cut ...]
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateDiversity(char, uint8_t, uint8_t)':
oled_128x64_adafruit_screens.cpp:498: error: 'map' was not declared in this scope
oled_128x64_adafruit_screens.cpp:500: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:500: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:503: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:508: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:516: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:521: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateSetupMenu(uint8_t, bool, bool, const char*, char)':
oled_128x64_adafruit_screens.cpp:534: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:534: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:536: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::save(uint8_t, uint8_t, uint16_t, const char*)':
oled_128x64_adafruit_screens.cpp:586: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:586: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:632: error: 'DEC' was not declared in this scope
oled_128x64_adafruit_screens.cpp: In member function 'void screens::updateSave(const char*)':
oled_128x64_adafruit_screens.cpp:649: error: 'display' was not declared in this scope
oled_128x64_adafruit_screens.cpp:649: error: 'WHITE' was not declared in this scope
oled_128x64_adafruit_screens.cpp:649: error: 'BLACK' was not declared in this scope
oled_128x64_adafruit_screens.cpp:650: error: 'strlen' was not declared in this scope

I am a newbie with Arduino, and the version is  Arduino 1.0.5-r2, could you please help me, Thanks.

[image]

jonno1130, 2016-01-12:
It looks like you havnt edited the adafruit oled library. You need to open the file in a text viewer and uncomment a few lines.

sheaivey, 2016-01-12:
You also need to update Arduino to the latest version... 1.6.7 and install Adafruit graphics library if you intend to use OLED.

Closing this issue as it is not a bug.

closed 2016-01-12 by sheaivey

xfce, 2016-01-12:
OK, thanks. After upgrade to  version 1.6.8, both TVOUT_SCREENS and OLED_128x64_ADAFRUIT_SCREENS compile successful