sheaivey/rx5808-pro-diversity #62: Add the following image for receiver version MM208R v1.3

sheaivey, 2016-01-28:
[](http://static.rcgroups.net/forums/attachments/6/1/8/3/9/5/a8646773-97-pin7.jpg)

chinoox, 2016-02-17:
Why this module is better?

sheaivey, 2016-02-17:
So far the original rx5808 chip has the fastest seek time and overall reliability as far as I can tell.

chinoox, 2016-02-17:
your project would be perfect if I had raceband.
how can i help you with raceband?

Ziege-One, 2016-02-17:
It is already in

5658, 5695, 5732, 5769, 5806, 5843, 5880, 5917 // Band C / Immersion Raceband

wzqvip, 2022-09-02:
By chance, I got a mm208r. But I can't find any kind of datasheet on the internet. 
It was from a G.T.power receiver, and only sells for ￥25， 
It is currently working in 8-ch mod, can it just replace the rx5808 module?