sheaivey/rx5808-pro-diversity #65: New L band

phoneflex, 2016-02-26:
it is possible to implement this new band?

Frequency Band L
CH1:5362
CH2:5399
CH3:5436
CH4:5473
CH5:5510
CH6:5547
CH7:5584
CH8:5621

yoannwyffels, 2016-03-02:
What is this "5.8ghz" band ? Frequencies are very far away from the 5.8ghz spectrum (5800mhz) and may be illegal in most countries around the world...

sheaivey, 2016-03-11:
Never heard of this L Band and @yoannwyffels makes a good point. Also I'm not sure these RX5808 modules can tune to those frequencies. Also good luck finding a transmitter that will operate in those frequencies.

That said the code change to send the command is pretty simple and you are welcome to run some experiments of your own and report back.

closed 2016-03-11 by sheaivey