sheaivey/rx5808-pro-diversity #69: i have some worng

wungsun, 2016-04-19:
> (D:\Arduino\Lib\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:389:6: error: prototype for 'void Adafruit_SSD1306::dim(boolean)' does not match any in class 'Adafruit_SSD1306'
> 
>  void Adafruit_SSD1306::dim(boolean dim) {
> 
>       ^
> 
> In file included from D:\Arduino\Lib\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:28:0:
> 
> D:\Arduino\Lib\libraries\Adafruit_SSD1306\Adafruit_SSD1306.h:142:8: error: candidate is: void Adafruit_SSD1306::dim(uint8_t)
> 
>    void dim(uint8_t contrast);

#  how to repair
my WeChat bi4iyj if you can.

sheaivey, 2016-05-23:
You may be missing the Adafruit libraries.

closed 2016-05-23 by sheaivey

wungsun, 2016-05-26:
Thank for reply！ I reinstalled all libraries。now is ok！

       祝好

王赏昌