sheaivey/rx5808-pro-diversity #71: prototype for 'char screens::begin()' does not match any in class 'screens'

vcazan, 2016-05-02:
Hey im trying to compile the non adafruit display version and I keep getting this error, any advise?

"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
Using previously compiled file: C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\TVOut_screens.cpp.o
Using previously compiled file: C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_adafruit_screens.cpp.o
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\Vlad\Documents\Arduino\libraries\U8glib\src" "-IC:\Users\Vlad\Documents\Arduino\libraries\TVout" "-IC:\Users\Vlad\Documents\Arduino\libraries\TVoutfonts" "C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp" -o "C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp.o"
oled_128x64_u8g_screens.cpp:53: error: prototype for 'char screens::begin()' does not match any in class 'screens'

 char screens::begin() {

      ^

In file included from C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp:39:0:

screens.h:50: error: candidate is: char screens::begin(const char*)

         char begin(const char *call_sign);

              ^

oled_128x64_u8g_screens.cpp:116: error: prototype for 'void screens::updateSeekMode(uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, bool)' does not match any in class 'screens'

 void screens::updateSeekMode(uint8_t state, uint8_t channelIndex, uint8_t channel, uint8_t rssi, uint16_t channelFrequency, bool locked) {

      ^

In file included from C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp:39:0:

screens.h:58: error: candidate is: void screens::updateSeekMode(uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, uint8_t, bool)

         void updateSeekMode(uint8_t state, uint8_t channelIndex, uint8_t channel, uint8_t rssi, uint16_t channelFrequency, uint8_t rssi_seek_threshold, bool locked); // seek and manual mode

              ^

C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp: In member function 'void screens::updateBandScanMode(bool, uint8_t, uint8_t, uint8_t, uint16_t, uint16_t, uint16_t)':

oled_128x64_u8g_screens.cpp:135: error: 'SCANNER_LIST_X_POS' was not declared in this scope

     static uint8_t writePos=SCANNER_LIST_X_POS;

                             ^

C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp: At global scope:

oled_128x64_u8g_screens.cpp:139: error: prototype for 'void screens::screenSaver(uint8_t, uint16_t)' does not match any in class 'screens'

 void screens::screenSaver(uint8_t channelName, uint16_t channelFrequency) {

      ^

In file included from C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp:39:0:

screens.h:66: error: candidates are: void screens::screenSaver(uint8_t, uint8_t, uint16_t, const char*)

         void screenSaver(uint8_t diversity_mode, uint8_t channelName, uint16_t channelFrequency, const char *call_sign);

              ^

screens.h:65: error: void screens::screenSaver(uint8_t, uint16_t, const char*)

         void screenSaver(uint8_t channelName, uint16_t channelFrequency, const char *call_sign);

              ^

oled_128x64_u8g_screens.cpp:142: error: prototype for 'void screens::screenSaver(uint8_t, uint8_t, uint16_t)' does not match any in class 'screens'

 void screens::screenSaver(uint8_t diversity_mode, uint8_t channelName, uint16_t channelFrequency) {

      ^

In file included from C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp:39:0:

screens.h:66: error: candidates are: void screens::screenSaver(uint8_t, uint8_t, uint16_t, const char*)

         void screenSaver(uint8_t diversity_mode, uint8_t channelName, uint16_t channelFrequency, const char *call_sign);

              ^

screens.h:65: error: void screens::screenSaver(uint8_t, uint16_t, const char*)

         void screenSaver(uint8_t channelName, uint16_t channelFrequency, const char *call_sign);

              ^

oled_128x64_u8g_screens.cpp:163: error: prototype for 'void screens::save(uint8_t, uint8_t, uint16_t)' does not match any in class 'screens'

 void screens::save(uint8_t mode, uint8_t channelIndex, uint16_t channelFrequency) {

      ^

In file included from C:\Users\Vlad\AppData\Local\Temp\build63c49c85898a371be4a35fbd0f3cefff.tmp\sketch\oled_128x64_u8g_screens.cpp:39:0:

screens.h:79: error: candidate is: void screens::save(uint8_t, uint8_t, uint16_t, const char*)

         void save(uint8_t mode, uint8_t channelIndex, uint16_t channelFrequency, const char *call_sign);

              ^

Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
Using library U8glib at version 1.19.1 in folder: C:\Users\Vlad\Documents\Arduino\libraries\U8glib 
Using library TVout at version 1.0.1 in folder: C:\Users\Vlad\Documents\Arduino\libraries\TVout 
Using library TVoutfonts in folder: C:\Users\Vlad\Documents\Arduino\libraries\TVoutfonts (legacy)
exit status 1
prototype for 'char screens::begin()' does not match any in class 'screens'
[image]

sheaivey, 2016-05-23:
u8g library was never completed due to library limitations.

closed 2016-05-23 by sheaivey

vcazan, 2016-05-23:
Good to know, might be worth taking it out from the codebase in master.