sheaivey/rx5808-pro-diversity #75: More channel table errors for raceband

Dvogonen, 2016-06-09:
There is already a bug report about the spi table error for raceband channel 5. The deviation leads to a huge frequency error in the RTC6715 frequenzy synthesizer (5967MHz set instead of the correct  5806MHz).

I am working on code to directly control the RitchWave RTC6715 circuit and checked my frequency calculation code against the spiValues table in this repository. It turns out that there are four additional errors in the raceband table that each result in a 2MHz frequency deviation. 2 MHz isn't that much, but it is better to be spot on than slightly off.

Current table:

const uint16_t channelTable[] PROGMEM = {
  // Channel 1 - 8
  0x2a05 ,0x299b ,0x2991 ,0x2987 ,0x291d ,0x2913 ,0x2909 ,0x289f ,
  0x2903 ,0x290c ,0x2916 ,0x291f ,0x2989 ,0x2992 ,0x299c ,0x2a05 ,
  0x2895 ,0x288b ,0x2881 ,0x2817 ,0x2a0f ,0x2a19 ,0x2a83 ,0x2a8d ,
  0x2906 ,0x2910 ,0x291a ,0x2984 ,0x298e ,0x2998 ,0x2a02 ,0x2a0c ,
  0x281d ,0x288f ,0x2902 ,0x2914 ,0x2978 ,0x2999 ,0x2a0c ,0x2a1e 
}

Corrected table

const uint16_t channelTable[] PROGMEM = {
  // Channel 1 - 8
  0x2a05 ,0x299b ,0x2991 ,0x2987 ,0x291d ,0x2913 ,0x2909 ,0x289f ,
  0x2903 ,0x290c ,0x2916 ,0x291f ,0x2989 ,0x2992 ,0x299c ,0x2a05 ,
  0x2895 ,0x288b ,0x2881 ,0x2817 ,0x2a0f ,0x2a19 ,0x2a83 ,0x2a8d ,
  0x2906 ,0x2910 ,0x291a ,0x2984 ,0x298e ,0x2998 ,0x2a02 ,0x2a0c ,
  0x281d ,0x2890 ,0x2902 ,0x2915 ,0x2987 ,0x299a ,0x2a0c ,0x2a1f 
}

pfriedel, 2016-06-10:
So as a cross check, I ran those numbers through the datasheet SPI mode register code (FLO = 2*(N*32+A)*(Fosc/R)) and I get the following tuned frequencies - "calc" is what the N and A values would tune in to, "tgt" is the target for that channel in that band, and whether it's off by one.  I believe the math prohibits from getting any closer than 1mhz off on even numbered frequencies - scanning the alternate tuning options shows they're all off by one.

<pre>
Band A:
hex    calc tgt
0x2a05 5865 5865
0x299b 5845 5845
0x2991 5825 5825
0x2987 5805 5805
0x291d 5785 5785
0x2913 5765 5765
0x2909 5745 5745
0x289f 5725 5725

Band B:
0x2903 5733 5732 off-by-one
0x290c 5751 5752 off-by-one
0x2916 5771 5771
0x291f 5789 5790 off-by-one
0x2989 5809 5809
0x2992 5827 5828 off-by-one
0x299c 5847 5847
0x2a05 5865 5866 off-by-one

Band E:
0x2895 5705 5705
0x288b 5685 5685
0x2881 5665 5665
0x2817 5645 5645
0x2a0f 5885 5885
0x2a19 5905 5905
0x2a83 5925 5925
0x2a8d 5945 5945

Band F:
0x2906 5739 5740 off-by-one
0x2910 5759 5760 off-by-one
0x291a 5779 5780 off-by-one
0x2984 5799 5800 off-by-one
0x298e 5819 5820 off-by-one
0x2998 5839 5840 off-by-one
0x2a02 5859 5860 off-by-one
0x2a0c 5879 5880 off-by-one

Band C / RB:
0x281d 5657 5658 off-by-one
0x2890 5695 5695 (original 0x288f is 5693mhz)
0x2902 5731 5732 off-by-one
0x2915 5769 5769 (original 0x2914 is 5767mhz)
0x2987 5805 5806 off-by-one (original 0x2978 is 5967mhz, I think?)
0x299a 5843 5843 (original 0x2999 is 5841mhz)
0x2a0c 5879 5880 off-by-one
0x2a1f 5917 5917 (original 0x2a1e is 5915mhz)
</pre>

sheaivey, 2016-06-10:
Thanks for the heads up @Dvogonen and @pfriedel. I'm actually looking into this now as well. The equation gives us float values (XXXX.5) however we can only send whole numbers and depending on the round this could leave us off by one or two MHz.

I put together this function a while back however I never fully trusted it to handle float conversions. Right now I'm giving it the full shake down to see how well it performs.
```
// calculate the frequency to bit bang payload
uint16_t freq_to_reg(uint16_t f)
{
    uint16_t tf, N, A;
    tf = (f - 479) / 2;
    N = tf / 32;
    A = tf % 32;
    return (N<<7) + A;
}
```

sheaivey, 2016-06-10:
Using my function i posted above I get about the same results as @pfriedel if I round the result.
```
// Band A
VM454:58 Freq: 5865, Calc: 0x2A05 (0x2A05) GOOD
VM454:58 Freq: 5845, Calc: 0x299B (0x299B) GOOD
VM454:58 Freq: 5825, Calc: 0x2991 (0x2991) GOOD
VM454:58 Freq: 5805, Calc: 0x2987 (0x2987) GOOD
VM454:58 Freq: 5785, Calc: 0x291D (0x291D) GOOD
VM454:58 Freq: 5765, Calc: 0x2913 (0x2913) GOOD
VM454:58 Freq: 5745, Calc: 0x2909 (0x2909) GOOD
VM454:58 Freq: 5725, Calc: 0x289F (0x289F) GOOD
VM454:51 
// Band B
VM454:58 Freq: 5733, Calc: 0x2903 (0x2903) GOOD
VM454:58 Freq: 5752, Calc: 0x290D (0x290C) OFF BY 1
VM454:58 Freq: 5771, Calc: 0x2916 (0x2916) GOOD
VM454:58 Freq: 5790, Calc: 0x2920 (0x291F) OFF BY 1
VM454:58 Freq: 5809, Calc: 0x2989 (0x2989) GOOD
VM454:58 Freq: 5828, Calc: 0x2993 (0x2992) OFF BY 1
VM454:58 Freq: 5847, Calc: 0x299C (0x299C) GOOD
VM454:58 Freq: 5866, Calc: 0x2A06 (0x2A05) OFF BY 1
VM454:51 
// Band E
VM454:58 Freq: 5705, Calc: 0x2895 (0x2895) GOOD
VM454:58 Freq: 5685, Calc: 0x288B (0x288B) GOOD
VM454:58 Freq: 5665, Calc: 0x2881 (0x2881) GOOD
VM454:58 Freq: 5645, Calc: 0x2817 (0x2817) GOOD
VM454:58 Freq: 5885, Calc: 0x2A0F (0x2A0F) GOOD
VM454:58 Freq: 5905, Calc: 0x2A19 (0x2A19) GOOD
VM454:58 Freq: 5925, Calc: 0x2A83 (0x2A83) GOOD
VM454:58 Freq: 5945, Calc: 0x2A8D (0x2A8D) GOOD
VM454:51 
// Band F / Airwave
VM454:58 Freq: 5740, Calc: 0x2907 (0x2906) OFF BY 1
VM454:58 Freq: 5760, Calc: 0x2911 (0x2910) OFF BY 1
VM454:58 Freq: 5780, Calc: 0x291B (0x291A) OFF BY 1
VM454:58 Freq: 5800, Calc: 0x2985 (0x2984) OFF BY 1
VM454:58 Freq: 5820, Calc: 0x298F (0x298E) OFF BY 1
VM454:58 Freq: 5840, Calc: 0x2999 (0x2998) OFF BY 1
VM454:58 Freq: 5860, Calc: 0x2A03 (0x2A02) OFF BY 1
VM454:58 Freq: 5880, Calc: 0x2A0D (0x2A0C) OFF BY 1
VM454:51 
// Band C / Immersion Raceband
VM454:58 Freq: 5658, Calc: 0x281E (0x281D) OFF BY 1
VM454:58 Freq: 5695, Calc: 0x2890 (0x288F) OFF BY 1
VM454:58 Freq: 5732, Calc: 0x2903 (0x2902) OFF BY 1
VM454:58 Freq: 5769, Calc: 0x2915 (0x2914) OFF BY 1
VM454:58 Freq: 5806, Calc: 0x2988 (0x2987) OFF BY 1
VM454:58 Freq: 5843, Calc: 0x299A (0x2999) OFF BY 1
VM454:58 Freq: 5880, Calc: 0x2A0D (0x2A0C) OFF BY 1
VM454:58 Freq: 5917, Calc: 0x2A1F (0x2A1E) OFF BY 1
```

If no rounding is done but instead just lob off the float I get the following results...
```
// Band A
VM456:58 Freq: 5865, Calc: 0x2A05 (0x2A05) GOOD
VM456:58 Freq: 5845, Calc: 0x299B (0x299B) GOOD
VM456:58 Freq: 5825, Calc: 0x2991 (0x2991) GOOD
VM456:58 Freq: 5805, Calc: 0x2987 (0x2987) GOOD
VM456:58 Freq: 5785, Calc: 0x291D (0x291D) GOOD
VM456:58 Freq: 5765, Calc: 0x2913 (0x2913) GOOD
VM456:58 Freq: 5745, Calc: 0x2909 (0x2909) GOOD
VM456:58 Freq: 5725, Calc: 0x289F (0x289F) GOOD
VM456:51 
// Band B
VM456:58 Freq: 5733, Calc: 0x2903 (0x2903) GOOD
VM456:58 Freq: 5752, Calc: 0x290C (0x290C) GOOD
VM456:58 Freq: 5771, Calc: 0x2916 (0x2916) GOOD
VM456:58 Freq: 5790, Calc: 0x291F (0x291F) GOOD
VM456:58 Freq: 5809, Calc: 0x2989 (0x2989) GOOD
VM456:58 Freq: 5828, Calc: 0x2992 (0x2992) GOOD
VM456:58 Freq: 5847, Calc: 0x299C (0x299C) GOOD
VM456:58 Freq: 5866, Calc: 0x2A05 (0x2A05) GOOD
VM456:51 
// Band E
VM456:58 Freq: 5705, Calc: 0x2895 (0x2895) GOOD
VM456:58 Freq: 5685, Calc: 0x288B (0x288B) GOOD
VM456:58 Freq: 5665, Calc: 0x2881 (0x2881) GOOD
VM456:58 Freq: 5645, Calc: 0x2817 (0x2817) GOOD
VM456:58 Freq: 5885, Calc: 0x2A0F (0x2A0F) GOOD
VM456:58 Freq: 5905, Calc: 0x2A19 (0x2A19) GOOD
VM456:58 Freq: 5925, Calc: 0x2A83 (0x2A83) GOOD
VM456:58 Freq: 5945, Calc: 0x2A8D (0x2A8D) GOOD
VM456:51 
// Band F / Airwave
VM456:58 Freq: 5740, Calc: 0x2906 (0x2906) GOOD
VM456:58 Freq: 5760, Calc: 0x2910 (0x2910) GOOD
VM456:58 Freq: 5780, Calc: 0x291A (0x291A) GOOD
VM456:58 Freq: 5800, Calc: 0x2984 (0x2984) GOOD
VM456:58 Freq: 5820, Calc: 0x298E (0x298E) GOOD
VM456:58 Freq: 5840, Calc: 0x2998 (0x2998) GOOD
VM456:58 Freq: 5860, Calc: 0x2A02 (0x2A02) GOOD
VM456:58 Freq: 5880, Calc: 0x2A0C (0x2A0C) GOOD
VM456:51 
// Band C / Immersion Raceband
VM456:58 Freq: 5658, Calc: 0x281D (0x281D) GOOD
VM456:58 Freq: 5695, Calc: 0x2890 (0x288F) OFF BY 1
VM456:58 Freq: 5732, Calc: 0x2902 (0x2902) GOOD
VM456:58 Freq: 5769, Calc: 0x2915 (0x2914) OFF BY 1
VM456:58 Freq: 5806, Calc: 0x2987 (0x2987) GOOD
VM456:58 Freq: 5843, Calc: 0x299A (0x2999) OFF BY 1
VM456:58 Freq: 5880, Calc: 0x2A0C (0x2A0C) GOOD
VM456:58 Freq: 5917, Calc: 0x2A1F (0x2A1E) OFF BY 1
```
Will need to do a bit more testing to make sure it rounded up vs down correctly.

sheaivey, 2016-06-10:
@pfriedel Do you mind adding your function snippet to convert from hex to frequency?

This is what I came up with but only works for about half the channels for some reason. (it must be way past my bed time :P)

```
uint16_t function reg_to_frq(uint16_t hex_freq)
{
    uint16_t N, A;
    A = (hex_freq & 0xf);
    N = (hex_freq >> 7);
    return (2 * (N * 32 + A)*(1))+479;
}
```

pfriedel, 2016-06-10:
Oh, mine was just a hack of a couple of perl for() loops throwing numbers into sprintf() and oct() to make sure the outputs looked appropriate.  I've attached it, but it's idiomatic as hell, just something I knocked out to verify @Dvogonen 's numbers. So some thoughts I had as I was falling asleep:

[numcount.pl.zip](https://github.com/sheaivey/rx5808-pro-diversity/files/309050/numcount.pl.zip)

1) There's that mystery meat "479" in the register calculation math.  I assume that's the IF on the module, since that's the only other 479 in the entire datasheet, and that's merely the minimum value, it could be 480.  Which means everything could be up 1mhz from the expected value? Yech.

2) That said, you could maybe set 0x00 (SYN_RF_R_REG) to 0x10 (up from the default 0x8) and get a by-two divisor, which gets you everything dead on with entirely different counter settings like...

(SYN is the synthesized number after the IF is added..)

<pre>
  N   A    N_BIN   A_BIN   DEC  HEX  SYN  TGT NOTES
158 110 10011110 1101110 20334 4f6e 5645 5645
158 123 10011110 1111011 20347 4f7b 5658 5658
159  78 10011111 1001110 20430 4fce 5645 5645
159  91 10011111 1011011 20443 4fdb 5658 5658
159  98 10011111 1100010 20450 4fe2 5665 5665
159 118 10011111 1110110 20470 4ff6 5685 5685
160  46 10100000 0101110 20526 502e 5645 5645
160  59 10100000 0111011 20539 503b 5658 5658
160  66 10100000 1000010 20546 5042 5665 5665
160  86 10100000 1010110 20566 5056 5685 5685
160  96 10100000 1100000 20576 5060 5695 5695
160 106 10100000 1101010 20586 506a 5705 5705
160 126 10100000 1111110 20606 507e 5725 5725
161  14 10100001 0001110 20622 508e 5645 5645
161  27 10100001 0011011 20635 509b 5658 5658
161  34 10100001 0100010 20642 50a2 5665 5665
161  54 10100001 0110110 20662 50b6 5685 5685
161  64 10100001 1000000 20672 50c0 5695 5695
161  74 10100001 1001010 20682 50ca 5705 5705
161  94 10100001 1011110 20702 50de 5725 5725
161 101 10100001 1100101 20709 50e5 5732 5732
161 102 10100001 1100110 20710 50e6 5733 5733
161 109 10100001 1101101 20717 50ed 5740 5740
161 114 10100001 1110010 20722 50f2 5745 5745
161 121 10100001 1111001 20729 50f9 5752 5752
162   2 10100010 0000010 20738 5102 5665 5665
162  22 10100010 0010110 20758 5116 5685 5685
162  32 10100010 0100000 20768 5120 5695 5695
162  42 10100010 0101010 20778 512a 5705 5705
162  62 10100010 0111110 20798 513e 5725 5725
162  69 10100010 1000101 20805 5145 5732 5732
162  70 10100010 1000110 20806 5146 5733 5733
162  77 10100010 1001101 20813 514d 5740 5740
162  82 10100010 1010010 20818 5152 5745 5745
162  89 10100010 1011001 20825 5159 5752 5752
162  97 10100010 1100001 20833 5161 5760 5760
162 102 10100010 1100110 20838 5166 5765 5765
162 106 10100010 1101010 20842 516a 5769 5769
162 108 10100010 1101100 20844 516c 5771 5771
162 117 10100010 1110101 20853 5175 5780 5780
162 122 10100010 1111010 20858 517a 5785 5785
162 127 10100010 1111111 20863 517f 5790 5790
163   0 10100011 0000000 20864 5180 5695 5695
163  10 10100011 0001010 20874 518a 5705 5705
163  30 10100011 0011110 20894 519e 5725 5725
163  37 10100011 0100101 20901 51a5 5732 5732
163  38 10100011 0100110 20902 51a6 5733 5733
163  45 10100011 0101101 20909 51ad 5740 5740
163  50 10100011 0110010 20914 51b2 5745 5745
163  57 10100011 0111001 20921 51b9 5752 5752
163  65 10100011 1000001 20929 51c1 5760 5760
163  70 10100011 1000110 20934 51c6 5765 5765
163  74 10100011 1001010 20938 51ca 5769 5769
163  76 10100011 1001100 20940 51cc 5771 5771
163  85 10100011 1010101 20949 51d5 5780 5780
163  90 10100011 1011010 20954 51da 5785 5785
163  95 10100011 1011111 20959 51df 5790 5790
163 105 10100011 1101001 20969 51e9 5800 5800
[... 78 lines cut ...]
167  43 10100111 0101011 21419 53ab 5866 5866
167  57 10100111 0111001 21433 53b9 5880 5880
167  57 10100111 0111001 21433 53b9 5880 5880
167  62 10100111 0111110 21438 53be 5885 5885
167  82 10100111 1010010 21458 53d2 5905 5905
167  94 10100111 1011110 21470 53de 5917 5917
167 102 10100111 1100110 21478 53e6 5925 5925
167 122 10100111 1111010 21498 53fa 5945 5945
168   5 10101000 0000101 21509 5405 5860 5860
168  10 10101000 0001010 21514 540a 5865 5865
168  11 10101000 0001011 21515 540b 5866 5866
168  25 10101000 0011001 21529 5419 5880 5880
168  25 10101000 0011001 21529 5419 5880 5880
168  30 10101000 0011110 21534 541e 5885 5885
168  50 10101000 0110010 21554 5432 5905 5905
168  62 10101000 0111110 21566 543e 5917 5917
168  70 10101000 1000110 21574 5446 5925 5925
168  90 10101000 1011010 21594 545a 5945 5945
169  18 10101001 0010010 21650 5492 5905 5905
169  30 10101001 0011110 21662 549e 5917 5917
169  38 10101001 0100110 21670 54a6 5925 5925
169  58 10101001 0111010 21690 54ba 5945 5945
170   6 10101010 0000110 21766 5506 5925 5925
170  26 10101010 0011010 21786 551a 5945 5945
</pre>

pfriedel, 2016-06-10:
And I just tested @Dvogonen's channelTable on my bench model (which uses FX158 modules vs rx5808, both use the RTC6715) and It tunes the RB channels just fine, notably that Band C's 5th channel works now.

sheaivey, 2016-06-11:
@pfriedel I was thinking the same thing... 

> 2) That said, you could maybe set 0x00 (SYN_RF_R_REG) to 0x10 (up from the default 0x8) and get a by-two divisor, which gets you everything dead on with entirely different counter settings like...

Initially I'm hesitant to make the change without a lot of testing (field testing that is). Correct me if I'm wrong but it looks like you have already generated all the correct HEX values for the new SYN_RF_R_REG set to 0x10. So testing can begin right away.

In the mean time I'm looking at a quick fix for just the affected channels in the current codebase. I just need to get my reverse function working to see which side of the target frequency it actually landed on from the rounding. This will make sure we are getting closer and not further from the target.

sheaivey, 2016-06-11:
OK final results with corrected my reverse lookup function. Results confirm my speculations that on the channels in question there is now way to hit them dead on. Making any changes will just land you on the other side of the target frequency.

Look below for an example of what I am talking about.
**chan** = target channel
**freq** = target frequency
**spi** = current binary command found in the code
**\-** = columns after this point are calculated values using my functions.
**rev_spi** = takes spi column and reverse looks up the frequency
**spi_syn** = takes the freq column and calculates the binary command
**freq_syn** = takes the spi_syn and reverse looks up the frequency

```
// Band A
chan	 freq	 spi	 -	 rev_spi	 spi_syn	 freq_syn	 notes
A1 		 5865 	 2A05 	 - 	 5865 		 2A05 		 5865 		 GOOD
A2 		 5845 	 299B 	 - 	 5845 		 299B 		 5845 		 GOOD
A3 		 5825 	 2991 	 - 	 5825 		 2991 		 5825 		 GOOD
A4 		 5805 	 2987 	 - 	 5805 		 2987 		 5805 		 GOOD
A5 		 5785 	 291D 	 - 	 5785 		 291D 		 5785 		 GOOD
A6 		 5765 	 2913 	 - 	 5765 		 2913 		 5765 		 GOOD
A7 		 5745 	 2909 	 - 	 5745 		 2909 		 5745 		 GOOD
A8 		 5725 	 289F 	 - 	 5725 		 289F 		 5725 		 GOOD

// Band B
chan	 freq	 spi	 -	 rev_spi	 spi_syn	 freq_syn	 notes
B1 		 5733 	 2903 	 - 	 5733 		 2903 		 5733 		 GOOD
B2 		 5752 	 290C 	 - 	 5751 		 290D 		 5753 		 freq!=freq_syn off by 1
B3 		 5771 	 2916 	 - 	 5771 		 2916 		 5771 		 GOOD
B4 		 5790 	 291F 	 - 	 5789 		 2920 		 5791 		 freq!=freq_syn off by 1
B5 		 5809 	 2989 	 - 	 5809 		 2989 		 5809 		 GOOD
B6 		 5828 	 2992 	 - 	 5827 		 2993 		 5829 		 freq!=freq_syn off by 1
B7 		 5847 	 299C 	 - 	 5847 		 299C 		 5847 		 GOOD
B8 		 5866 	 2A05 	 - 	 5865 		 2A06 		 5867 		 freq!=freq_syn off by 1

// Band E
chan	 freq	 spi	 -	 rev_spi	 spi_syn	 freq_syn	 notes
E1 		 5705 	 2895 	 - 	 5705 		 2895 		 5705 		 GOOD
E2 		 5685 	 288B 	 - 	 5685 		 288B 		 5685 		 GOOD
E3 		 5665 	 2881 	 - 	 5665 		 2881 		 5665 		 GOOD
E4 		 5645 	 2817 	 - 	 5645 		 2817 		 5645 		 GOOD
E5 		 5885 	 2A0F 	 - 	 5885 		 2A0F 		 5885 		 GOOD
E6 		 5905 	 2A19 	 - 	 5905 		 2A19 		 5905 		 GOOD
E7 		 5925 	 2A83 	 - 	 5925 		 2A83 		 5925 		 GOOD
E8 		 5945 	 2A8D 	 - 	 5945 		 2A8D 		 5945 		 GOOD

// Band F / Airwave
chan	 freq	 spi	 -	 rev_spi	 spi_syn	 freq_syn	 notes
F1 		 5740 	 2906 	 - 	 5739 		 2907 		 5741 		 freq!=freq_syn off by 1
F2 		 5760 	 2910 	 - 	 5759 		 2911 		 5761 		 freq!=freq_syn off by 1
F3 		 5780 	 291A 	 - 	 5779 		 291B 		 5781 		 freq!=freq_syn off by 1
F4 		 5800 	 2984 	 - 	 5799 		 2985 		 5801 		 freq!=freq_syn off by 1
F5 		 5820 	 298E 	 - 	 5819 		 298F 		 5821 		 freq!=freq_syn off by 1
F6 		 5840 	 2998 	 - 	 5839 		 2999 		 5841 		 freq!=freq_syn off by 1
F7 		 5860 	 2A02 	 - 	 5859 		 2A03 		 5861 		 freq!=freq_syn off by 1
F8 		 5880 	 2A0C 	 - 	 5879 		 2A0D 		 5881 		 freq!=freq_syn off by 1

// Band C / Immersion Raceband
chan	 freq	 spi	 -	 rev_spi	 spi_syn	 freq_syn	 notes
C1 		 5658 	 281D 	 - 	 5657 		 281E 		 5659 		 freq!=freq_syn off by 1
C2 		 5695 	 288F 	 - 	 5693 		 2890 		 5695 		 GOOD
C3 		 5732 	 2902 	 - 	 5731 		 2903 		 5733 		 freq!=freq_syn off by 1
C4 		 5769 	 2914 	 - 	 5767 		 2915 		 5769 		 GOOD
C5 		 5806 	 2987 	 - 	 5805 		 2988 		 5807 		 freq!=freq_syn off by 1
C6 		 5843 	 2999 	 - 	 5841 		 299A 		 5843 		 GOOD
C7 		 5880 	 2A0C 	 - 	 5879 		 2A0D 		 5881 		 freq!=freq_syn off by 1
C8 		 5917 	 2A1E 	 - 	 5915 		 2A1F 		 5917 		 GOOD
```

sheaivey, 2016-06-11:
I think there is no real issue here with the current code as it stand. 

Our best course of action to hit the exact frequency is to pursue @pfriedel suggestion about changing the SYN_RF_R_REG command to 0x10 which will give us a more precise tune.

sheaivey, 2016-06-18:
Does not appear that you can set the SYN_RF_R_REG. Register A seems to revert back to default. Not sure what I'm doing wrong.

I created a helper function that sets a register. Give it the (address, data) and it bit bangs out the command.
Function is working because I have tested it with setting Register B (0x01) which is the frequency data.

This is the command i'm trying to send to Register A (0x00)
```
B0000 - address
B1 - write
B00000000000000010000 - data 0x10 ... default is 0x08
```

I also looked at Register C (0x02) thinking that I needed to change MOUT which is the "Multi-function output select" (RF R divider output, RF prescaler output, lock in detect)
But I was still unable to get change in operation. 

Maybe some one out there has had success setting other Registers or would like to take a try at it. @Dvogonen @pfriedel ??

Dvogonen, 2016-06-19:
Hi, I just checked back in to see how things where moving here.
I notice that you are trying to increase the precision of the receiver by changing the frequency divisor in SYN_RF-R_REG. I thought of that myself, but gave up on the idea since I came to the conclusion that it would not work. The frequency calculation according to the data sheet:
2(N*32+A)*(Fosc/R) + 479
The problem is that all calculations are on integers, not floats.
Whichever way you change N, A and R the result of (N*32+A)(Fosc/R) can only become an integer. This is then multiplied with with 2 and 479 is added.
The circuit is simply not capable of using even frequencies. You will always end up with an odd result.

sheaivey, 2016-06-24:
Your assumption of the TX having the same issue is incorrect. The equation to set the channel on the VTX deals in KHz without a base frequency. This means the transmitter is capable of having a much finer tune and will land on even or odd without floating point.

```
FRF = 2*(N*64+A)*(Fosc/R)
FRF=5865MHz, Fosc=8MHz, R=400
￼5865=(2*N*64+A)*8Mhz/400=2*(N*64+A)*20
N=2291(=8F3xH), A=1(=01xH)
```

---

Back to the RX side of things if we were able to set the SYN_RF_R_REG to 0x10 (16) we would be able to hit the frequency dead on... even or odd. this is because the 0.5 floating point calculations happen to land us exactly on whole numbers when the frequency is even.

```
2*(N*32+A)*(Fosc/R) + 479
2*(N*32+A)*(8/16) + 479
2*(N*32+A)*(.5) + 479

// Odd example
N=168, A=10
2*(168*32+10)*(.5) + 479 = 5865

// Even example
N=168, A=11
2*(168*32+11)*(.5) + 479 = 5866
```

I reached out to the company who makes the chip and they suggested using 480 as the base frequency but just reverses the problem making us only able to land on even tune, making odd frequencies off by 1.

We are so close it would be unfortunate to give up now. **The only problem I am having is sending the command to set SYN_RF_R_REG to 0x10 (16)**

Dvogonen, 2016-06-24:
You are right. My mistake was that I did not think of the fact that the result of the frequency division will not become 0.5. It is instead 500000 since the circuit is dividing 8MHz with 16 and get 0.5MHz=500000Hz. And the division is not really a mathematical division, it is an instruction to ignore a specified number of clock ticks. 

I have got no really good idea why you can not set the divider to 0x10. The data sheet is very hard to understand. First it says that 0x10 is the default, then it goes on to say that 0x08 is the default. Strange. 

Have you tried writing the register several consecutive times? Or writing and reading the register in cycle a number of times ? The reason I ask is that I think I have seen intermittent failures to set channel values. The failures are not repeatable and only seem to occur the first time a channel value is written. The following writes succeed. But this is only speculations. I have not verified my observations.

sheaivey, 2016-06-25:
No worries @Dvogonen I have had a hard time trying to understand the data sheet as well. I'm going to take your suggestion and try sending the command a few times back to back. Hopefully that is all the issue is. I'll also double check all my calculations and make sure I'm sending everything right.

Side note, I was looking over your CYCLOP+ project and like where you are going with it. Should be a good improvement for the Quantums.

Dvogonen, 2016-06-25:
I am traveling and do not have access to my equipment, otherwise I would have made an attempt at writing the register out of curiosity. 

I will anyhow write code to verify that writing the channel setting register work. I have only seen what I suspect where failed writes twice. Hopefully I am wrong.

Thank you for your kind words on CYCLOP+. Using your project as inspiration and a source of information made an extreme difference when I wrote the CYCLOP+ code.

sheaivey, 2016-08-01:
This issue is deeper within the RTC6705 chips protocol not accepting certain SPI commands.

closed 2016-08-01 by sheaivey