sheaivey/rx5808-pro-diversity #77: rx5808 pro dual diversity  with the nano

buckrogers1, 2016-07-25:
Hi everyone . Im an ex missile tech. ,and am building the receiver, rx5808 pro dual diversity  with the nano .Can you guys connect me with build pictures? Ive been looking for two weeks and I cant seem to figure out the Git Hub .I got the boards from ohpark, have the nanos and all other parts . I want to build the model without the oled ,ill use a regular monitor . A picture of the top and bottom before and after the nano is installed . How long of extensions can the antennae connectors have on them for base station mounting before signal loss . When we first turn this on , do I need to paste and include some code into the arduino sketch library? All I read for directions were the parts list ,something about removing a part on each rx , and something about calibrating signal strength . There is code stuff all over but nothing seemingly directed at the nano version. It sure would be nice just to follow a recipe ,build the board , set up this ,do that, no you ready to go . Any help out there ?

zanthor, 2016-07-25:
> How long of extensions can the antennae connectors have on them for base
station mounting before signal loss.

Not long at all!  Anything over 6" and you start losing signal quickly.

buckrogers1, 2016-07-25:
Well thanks but it makes it kind of tough for if you want to hide cables for external antennae . I have the nano board version dual rx5808 and the parts list doesnt match the schematic and neither match the board  i got 3 of .My board has a group of resistors mounted on opposite side of board from nano ,2 leds. the nano has to be mounted on opposite side of board from rx to match pinout  i think . And Im also looking at the 15 db gain antennaes ,thinking this might cure the signal loss problem . This is all from the sheaivy site but Im too new to be able to find simple procedures for this or its not ther and not supposed to be . A little please . Oh by the way , my name is ken

sheaivey, 2016-07-25:
@buckrogers1 I'm not sure which board you are referring to? None of the purple boards from OshPark use the Arduino Nano.

The through-hole version boards use an Arduino Pro Mini @ 5v. you can find documentation on those boards here.
https://github.com/sheaivey/rx5808-pro-diversity/blob/master/docs/diy-through-hole-board.md

The only board that supports GUI Video Out to a monitor is the custom advanced board. 
https://github.com/sheaivey/rx5808-pro-diversity/blob/master/docs/diy-custom-board.md

Since it sounds like you are new to the Arduino world I suggest brushing up on some of the basics centered around uploading firmware to the MCU. You can also read more about this project over at the RCGroups thread.
http://www.rcgroups.com/forums/showthread.php?t=2477752

You are more then welcome to submit a pull request with updated documentation once you get things running.

-Shea

buckrogers1, 2016-07-25:
diy simple arduino nano rx 5808 pro diversity ,Im looking at it now in another window . yes I do need to learn arduino and i will . . Did I get the wrong board  for a nano build ?It has a place for two leds by the antennae jacks , 15 resistors in a row between  a 19 dip and room for 2 rx5808, and a 4066 chip. What do I have ?Id like to go nanno .Im pretty sure i ordered this as a nanno board Im sure the documentation said it was the same as another board except it uses a nanno .I will try to find that comment . Meanwhile I need to sort this out and maybe i messed up but well see .

buckrogers1, 2016-07-25:
https://oshpark.com/shared_projects/Y2Zlih1f       This is the one and my apoligies for being a rookie but i really want to build this so help me out if you can .

buckrogers1, 2016-07-26:
rx5808 Nano Board copy

 Top Bottom
2 layer board of 3.94x1.54 inches (100.00x39.01 mm). 
Uploaded on July 13th, 2016 18:02. 
Previous orders:

Ordered on July 13th, 2016 (SHIPPED) View order.
Half the height, all the fun. Ok, only 15mm out of 54, but it’s an important 15mm. Only change to the BOM is an Arduino Nano replacing a Pro Micro. Some quality of life improvements to the layout, but nothing significant.

buckrogers1, 2016-07-26:
OSH Park ~ rx5808 Nano Board
  
|  
|   
|   
|   |    |

   |

  |
|  
|   |  
OSH Park ~ rx5808 Nano Board
   |   |

  |

  |

 rx5808 Nano Board copyTop Bottom
2 layer board of 3.94x1.54 inches (100.00x39.01 mm). 
Uploaded on July 13th, 2016 18:02. 
Previous orders:Ordered on July 13th, 2016 (SHIPPED) View order.
Half the height, all the fun. Ok, only 15mm out of 54, but it’s an important 15mm. Only change to the BOM is an Arduino Nano replacing a Pro Micro. Some quality of life improvements to the layout, but nothing significant. this is from oh park shared projects and the board is a red mixed with blue closely resembling "purple" 

    On Monday, July 25, 2016 4:01 PM, Kenneth l Wagner <kennethlwagner@yahoo.com> wrote:
 

 diy simple arduino nano rx 5808 pro diversity ,Im looking at it now in another window . yes I do need to learn arduino and i will . . Did I get the wrong board  for a nano build ?It has a place for two leds by the antennae jacks , 15 resistors in a row between  a 19 dip and room for 2 rx5808, and a 4066 chip. What do I have ?Id like to go nanno .Im pretty sure i ordered this as a nanno board Im sure the documentation said it was the same as another board except it uses a nanno .I will try to find that comment . Meanwhile I need to sort this out and maybe i messed up but well see .

sheaivey, 2016-07-26:
@buckrogers1 I have not seen that board before but it is using the Arduino Nano. The nano and 4066 chips are mounted on the opposite side as the RX modules. Aside from using through hole components the build will be the same as the rest of the boards. I would suggest looking for more information on this board from where ever you found it from.

That is about all I can help with. When you finish I suggest creating a new documentation page and list out the build procedure you had to do to get this board working. Then do a PR to add it to this repo so others can have a better starting point.

-Shea

buckrogers1, 2016-07-27:
Ok ,So oh park lists resisitor as the person , my  lead .I will find the documentation if its there , maybe. But if its almost the same except for the nano substitution ,then I just have to include the code files into the arduino sketch library ,and maybe modify the code depending which model ,then check it with sketch, then upload to board . Right? I have to make a repository for my sketch to put copies of the masters you guys make in.  Waitin for the rain ,im an organic mushroom picker from oregon

closed 2016-12-13 by sheaivey