sheaivey/rx5808-pro-diversity #78: rx5808  dual diversity nano board

buckrogers1, 2016-07-29:
https://oshpark.com/shared_projects/Y2Zlih1f        I have this board version , of the rx 5808 dual diversity receiver using the nano . I have contacted osh park , sheaivy ,patrick , pfriedel and the closet i got was a red board that had some things in common with my version but not the same build at all . I have the purple rx 5808 nano board version that I had to purchase 3 of , and Ive been putting a couple hours a day for 2 weks now into finding the correct build info , no luck . Can you guys find the correct build docs. for the osh park nano board and send me a link please . Im rather new to github and rc forum  but someone else should be able to find this easier than me . Youd think osh park having sold me the boards from their shared projects would be able to connect me with a thousand other people but not yet .

buckrogers1, 2016-07-30:
https://cloud.githubusercontent.com/assets/20030826/17267545/e27ed14c-55c0-11e6-8873-4f28722bef9e.jpg
https://cloud.githubusercontent.com/assets/20030826/17267565/47f9c554-55c1-11e6-8ba1-0337250ea5ec.jpg

buckrogers1, 2016-07-30:
This what i have so far ,dont know where this is really supposed to go but its something

sheaivey, 2016-07-30:
Looks like you are well on your way! Keep it up. From what I have heard in the forums it is an pretty much the same as all the other throughhole builds.

sheaivey, 2016-08-01:
closing this issue as it is a duplicate of issue #77 
https://github.com/sheaivey/rx5808-pro-diversity/issues/77

closed 2016-08-01 by sheaivey