sheaivey/rx5808-pro-diversity #83: Add the ne L-Band (5362 MHz to 5621 MHz)

ps915, 2016-08-21:
Is it possible to add the new L-Band to use this new 48CH [VTx](http://www.banggood.com/FT48X-5_8G-48CH-25200600mW-Adjustable-Video-Transmitter-FPV-Racer-VTX-Raceband-p-1073899.html?p=3R26141006882201412N)

The new Band has the following channels:
5362 MHz 5399 MHz 5436 MHz 5473 MHz 5510 MHz 5547 MHz 5584 MHz 5621 MHz

[image]

Would be awesome, because then we have 16 raceband channels which are 37Mhz away from each other

asficas, 2016-09-02:
Done, enjoy!

ps915, 2016-09-03:
Great thanks so much, but you missed the "," buddy =)

[image]

What about the band Scanner and Auto Seek? Is it a full implementation?
Just flashed it and it seems working =)

asficas, 2016-09-03:
Ohh.. 3 times in fact! Made the corrections just now.
Yes it is a full implementation.
Thank you for starting this great project, I flew my quad yesterday on 5362!
Cheers.

ps915, 2016-09-04:
Thanks =) Well done!

sheaivey, 2016-09-10:
This has been merged into master.

closed 2016-09-10 by sheaivey

vadimus-ca, 2016-12-08:
Looks like the frequencies used for L band don't match the real products...

[image]

http://www.banggood.com/Eachine-TX03-Super-Mini-025mW50mW200mW-Switchable-AIO-5_8G-72CH-VTX-600TVL-13-Cmos-FPV-Camera-p-1104884.html

mdmbc, 2016-12-08:
This is probably going to keep happening as long as we keep relying on the non-standard band names vs frequency.

vadimus-ca, 2016-12-08:
True, yet already 2 products have L band listed there.

And 48ch VTX product page used as reference for L band now in code actually does not really claim it is L band.

Is there any online calc so I can encode needed SPI values for those missing bands U,O,L,H?

vadimus-ca, 2016-12-08:
Never mind, used sheaivey's function to calculate needed hex values for L and U bands, will try to test it once my 72ch VTX is delivered.

Someone might ask why do we care about those extra channels? The idea is to spread as much channels over the spectrum as far as you can so you can have a bigger number of pilots flying/racing same time.
Raceband has 8ch spaced by 37MHz and it was the best approach so far but with L and U combined we have 16ch spaced by 40MHz. 
[image]

vadimus-ca, 2016-12-09:
Tested these SPI values with 72ch VTX on L band.

All channels are fine.
[image]

tipouic, 2016-12-09:
HI @vadimus-ca,I'm very interested for your work because I have the same camera. Can you publish your code and the calculation file in order to add the rest of the frequencies?

vadimus-ca, 2016-12-09:
Sure, here it is: https://docs.google.com/spreadsheets/d/13k_F92v-d6mQLr359WKGDGbnlNQcZMJ6ZbLwvM2zsWU/edit?usp=sharing 
[image]

vadimus-ca, 2016-12-09:
`/*
 * SPI driver based on fs_skyrf_58g-main.c Written by Simon Chambers
 * TVOUT by Myles Metzel
 * Scanner by Johan Hermen
 * Inital 2 Button version by Peter (pete1990)
 * Refactored and GUI reworked by Marko Hoepken
 * Universal version my Marko Hoepken
 * Diversity Receiver Mode and GUI improvements by Shea Ivey
 * OLED Version by Shea Ivey
 * Seperating display concerns by Shea Ivey
 * 10 Favorite channels and functionality by George Chatzisavvidis
 * change in displays by George Chatzisavvidis
 * Adding Fatshark button to work with the module by George Chatzisavvidis

 * The MIT License (MIT)

Copyright (c) 2015 Marko Hoepken

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
*/

#include <avr/pgmspace.h>
#include <EEPROM.h>

#include "settings.h"

// uncomment depending on the display you are using.
// this is an issue with the arduino preprocessor
 
#ifdef OLED_128x64_ADAFRUIT_SCREENS

  #ifdef SH1106
  // #include <Adafruit_SH1106.h>
 #else
    #include <Adafruit_SSD1306.h>
 #endif
    #include <Adafruit_GFX.h>
    #include <Wire.h>
    #include <SPI.h>
#endif
//#ifdef OLED_128x64_U8G_SCREENS
//    #include <U8glib.h>
//#endif
#include "screens.h"
screens drawScreen;

#define EEPROM_ADR_TUNE_FAV_LAST 108

int EEPROM_ADR_TUNE_FAV[10]  = {100, 101, 102, 103, 104, 105, 106,107,110,111};
int temp_EEPROM_ADR_TUNE_FAV[10] = {255, 255, 255, 255, 255, 255, 255,255,255,255};
//setting the 10 favorite channels memory blocks George Chatzisavvidis GC9N

// Channels to sent to the SPI registers
const uint16_t channelTable[] PROGMEM = {
[... 1457 lines cut ...]
void SERIAL_ENABLE_LOW()
{
  delayMicroseconds(1);
  digitalWrite(slaveSelectPin, LOW);
  delayMicroseconds(1);
}

void SERIAL_ENABLE_HIGH()
{
  delayMicroseconds(1);
  digitalWrite(slaveSelectPin, HIGH);
  delayMicroseconds(1);
}

void FillTemp_Tune_fav(int FavChannelx) {
        for(int i = 0; i<10; i++)
                {
                    if ( temp_EEPROM_ADR_TUNE_FAV[i] == 255 && FavChannelx != 255) //not used  gc9n
                    {
                      temp_EEPROM_ADR_TUNE_FAV[i]=FavChannelx;
                      i=12;
                    }
                   
                }
}`

vadimus-ca, 2016-12-09:
[rx5808-pro-diversityGc9n-Lbandcorrected.zip](https://github.com/sheaivey/rx5808-pro-diversity/files/642923/rx5808-pro-diversityGc9n-Lbandcorrected.zip)

vadimus-ca, 2016-12-09:
To get full list of frequencies some logic should be revised in the code (frequency range in histogram drawing, band names etc) - a bit too big piece of work for me at the moment.

tipouic, 2016-12-11:
Thank you, sorry for the time of my answer because I work a little on the code in my fork to do the integration of the 72 channels.
[https://github.com/tipouic/rx5808-pro-diversity](https://github.com/tipouic/rx5808-pro-diversity)

tipouic, 2016-12-17:
Hi @vadimus-ca , I try to put also these new bands in the project of the emitter but I can not find the solution (because the hex values are different), by chance you would not know where I can find it?

vadimus-ca, 2016-12-17:
What are you looking for exactly?

tipouic, 2016-12-17:
The transmitter which with this receiver [https://github.com/sheaivey/tx5823-pro](url)

Gintaris, 2017-03-01:
Hi @vadimus-ca , today I've got Eachine VTX03 , on L band just first channel doesn't work, all other are working corectly, maybe there is some mis calculation in SPI values ?