sheaivey/rx5808-pro-diversity #84: Control module via Serial

tgreer, 2016-08-22:
Hi Guys

i'm looking to use these modules in a project i'm working on: 
http://www.banggood.com/Realacc-RX5808-Open-Source-5_8G-40CH-Integrated-Receiver-with-OLED-Special-for-Fatshark-Goggles-p-1068200.html

I currently use TBS modules, and I can control them via serial (I have an arduino hooked up to tell them which channel to tune to) which talks via the channel updown pins on the module header

Do you know if it's possible to do that with the existing firmware, and if not, where i'd need to look at changing things in order to make this possible?

Cheers

Thomas

derFrickler, 2016-09-29:
if you can program arduino this should not be a big thing. 

If you dont't need the Oled Menu you can remove a lot of code and just listen on the serial for your commands, then call setChannelModule(uint8_t channel) with the channel you want to tune to.
Where channel is the index of the wanted channel in the channel table at the top of the sketch
https://github.com/sheaivey/rx5808-pro-diversity/blob/master/src/rx5808-pro-diversity/rx5808-pro-diversity.ino#L65

derFrickler, 2016-10-04:
Check out that project, the example does what you need: 
https://github.com/Spacefish/RichWave

sheaivey, 2016-10-10:
I think @derFrickler listed the best solution to this issue.

closed 2016-10-10 by sheaivey