sheaivey/rx5808-pro-diversity #87: Favourites Channel List

ps915, 2016-09-04:
What about adding a Favourites Channel List?

I have different Quads on different channels and want to quick swap between them.

Would be great! =)

kabturek, 2016-10-10:
@sheaivey is it ok if i mimic the look & feel of the fav menu of the la forge or should i create something different ?

sheaivey, 2016-10-10:
I would suggest coming up with your own way of representing and managing favorites. You may come up with a more elegant solution then what I came up with for the La Forge FPV firmware.

gc9n, 2016-10-11:
i also gonna make this menu its a must have feature! , kabturek did you start the development? 
cause i already started the code..

kabturek, 2016-10-11:
@sheaivey thanks, will try
@gc9n good to hear, i've implemented a basic fav menu yesterday (for the tvout screen) but not tested it yet on the actual device as i dont have it with me

gc9n, 2016-10-11:
i want to help as far as i can to make this good. 
Do you guys use a diy hardware ? because right now i have the Banggood version?
When you say TVOut you mean OLED UI ? correct?

kabturek, 2016-10-11:
yes i have a diy hardware (on boards from the wiki). I have the receiver built into a screen so im using the tv out to output the interface to one of the screens inputs - so im testing on the "TVout" screen ( its called screens in source). so it's not the OLED UI but i plan on porting it to OLED when i'm finished. 

if you have any ideas about the interface please feel free to share :)

gc9n, 2016-10-11:
What about an esp8266 and send data over and android?

gc9n, 2016-10-11:
any way i have the banggood version cause i plugged it into a fatshark HDV2

Knifa, 2016-12-10:
Looking into this as well. Currently trying to do a refactor to make adding new menus and screens easier.