sheaivey/rx5808-pro-diversity #89: Can't compile for TVout

neirbomit, 2016-09-27:
Hi,
this is the first time posting an issue so hopefully I explain myself in a way that works for everyone.
I've been trying to compile rx5808-pro-diversity for TVout for a couple of days on and off now and have had no success. I got the OLED version compiling (although I don't have a screen so I can't test it). I get warnings about use of memory but other than that all good.  As for the TV out I have included the appropriate libraries (I think) but I get these 

Error: unknown opcode `svprt' 

for all of the macros in the asm_macros.h file (in TVout/spec). If I try to add #include "spec\asm_macros.h" I just get

Error: Macro `svprt' was already defined

I had been installing a bunch of other GCC libraries for other IDE's and I figured it was possible something got pointed to the wrong library so I ulgraded my arduino from 1.6.11 to 1.6.12 and still ended up with the same errors.  I'm at a loss. Does anybody have any ideas?

Here is a complete listing of the errors I am getting. I can include the verbose version if someone thinks that would be usefull.

Thanks:)

Arduino: 1.6.12 (Windows 7), Board: "Arduino Nano, ATmega328"

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s: Assembler messages:

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:555: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:594: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:596: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:599: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:602: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:605: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:608: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:611: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:616: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:619: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:652: Error: unknown opcode `svprt'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:656: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:660: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:661: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:663: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:664: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:666: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:667: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:669: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:670: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:672: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:673: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:675: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:676: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:679: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:681: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:683: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:684: Error: unknown opcode `svprt'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:686: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:718: Error: unknown opcode `svprt'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:722: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:725: Error: unknown opcode `delay1'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:727: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:728: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:730: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:731: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:733: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:734: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:736: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:737: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:739: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:740: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:742: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:743: Error: unknown opcode `delay3'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:745: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:748: Error: unknown opcode `delay2'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:750: Error: unknown opcode `o1bs'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:751: Error: unknown opcode `svprt'

C:\Users\Tim\AppData\Local\Temp\ccGSXta1.s:753: Error: unknown opcode `o1bs'

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ps915, 2016-09-27:
I was also not able to compile it on tte latest Arduino IDE. 

For me the Version 1.6.8 is working without Errors

http://blog.seidel-philipp.de/rx5808-diversity-modul-flashen-laforge-true-d-realacc/#Software

neirbomit, 2016-09-28:
Thank you, that seemed to work (in that it compiles now, I have yet to test functionality or burn it to a board). I'm wondering if the problem is a bug in the new arduino code or just incompatibility or if there is something wrong with my system configuration. It's strange though that the TVout demo projects compiled just fine for me.  Hopefully I won't have any more issues.

sheaivey, 2016-10-10:
This appears to be a non issue.

closed 2016-10-10 by sheaivey

neirbomit, 2016-10-10:
Only as long as you use Arduino 1.6.8. Later versions do not compile.

akua73, 2016-10-14:
Hello!
Please... pls..Any solution to this erratic behaviour ? I can compile samples of tvout in 1.6.12 but sometimes my own code does not compile showing the same opcodes messages :(
Thks!!!!

neirbomit, 2016-10-14:
Hey akua73, 
I was able to compile TVout too but not the rx5808 code I don't know where the problem is but try using version 1.6.8 as that totally fixed the problem with me.  I'm assuming it messes up something when certain combinations of libraries mess up the compiler or linker with some change they've made in Arduino. I don't know how or why but hopefully version 1.6.8 fixes your code too.  It really sucks to be stuck using an old version but maybe they'll fix it at some point.

akua73, 2016-10-14:
thanks a lot for your answer neirbomit,
yes, my code using tv-out library (this one : https://github.com/Avamander/arduino-tvout ) in a pro-micro 5v/16mhz and UI 1.6.8 compiles well, but the result in my tv screen is not stable, the image is erratic ... also the examples in PAL or NTSC : the image in the TV screen is not stable ... :(
so I have two options : 
1- try to solve the opcode errors in UI 1.16.12
2- try to solve the unstable tv-out screen using UI 1.16.8
and , if possible , it seems better the option1 :)
don’t know,, i will try to contact arduino developers... and will see... 
any other ideas will be very appreciated ! 
thanks!!
PS: on the other hand,, in tv-out the file video_gen.cpp seems to have all the assembler code that could be causing these errors when compiling, I will take a look here …

Mickyleitor, 2016-10-14:
Hi,
Same happens here in 1.6.12. The same opcode show up using the TV-out library. It's weird as the sample code compile, and as soon as I put more code (even not "playing" with the TV) then It will through those errors..

neirbomit, 2016-10-14:
akua73 - I wonder if the stability is from the version or some other timing thing in your code? 1.6.8 is perfectly stable for me. Is it a possible hard ware problem, loose connection,  wrong resistor value? Maybe even check your actual monitor input with a different source.

Mickyleitor - I just tried to add a bunch of random code to TV-out in 1.6.12 and it compiled for me. Is there a specific library that you added or specific piece of code that you added when it stopped compiling? Maybe try and compile for a different board?

Mickyleitor, 2016-10-14:
@neirbomit I'm using TVout.h and fontALL.h, I will add more in the future but It's ok for debug purposes. The code is using the standard serial to read and print from/to the keyboard/monitor (I know, this will stop momentany the TV streaming but not really a problem for my application). 
I will try to take out different parts of code finding out what code could be.

neirbomit, 2016-10-14:
@akua73, Mickeyleitor just gave me an idea for your stability problem. It might not be the issue but Serial.print statements are blocking in arduino if there is too much data for the buffer.  If you are using a lot of print statements you might want to consider commenting them out to see if this fixes you problem.  If it does you might want to check the buffer flags before writing anything to serial. Or maybe even switching to a much higher serial rate.

akua73, 2016-10-14:
I think i already tested it (remove all serial debug prints)... Dont remember .... On the other hand I can not test anything by now... But  I will try all the ideas in my next experiments session... Thanks for the help

akua73, 2016-10-17:
Now i see that all these problems regarding macros were already discovered in the first post ;) ... Sorry!
Maybe we can just substitute the code in the macros directly in video_gen.cpp .... Dont know...

neirbomit, 2016-10-17:
@akua73 It's really just weird. If you look at the macros that (delay2 for example) that's all it's doing, it should just be replacing the itself with a couple of nops before being compiled. Something in the linker has got to be messed up, but why sometime and not others and why with the newer versions.  I'll have to look over the code again. I'm not sure what #define REMOVE6C etc do. that's a step in the right direction though... maybe.

akua73, 2016-10-17:
well, at the end I achieved , more or less, the goal : it compiles and it works (in 1.6.12)... but yes.. very extrange,,, i dont know why the macros are not running properly ...
the solution was simply to substitue delay1,delay2,delay3, o1bs and svprt code directly ...

I used this sed file if someone is interested in doing the same workaround:

```
s/"delay3\\n\\t"/"nop\\n\\t"\n"nop\\n\\t"\n"nop\\n\\t"/g
s/"delay2\\n\\t"/"nop\\n\\t"\n"nop\\n\\t"/g
s/"delay1\\n\\t"/"nop\\n\\t"/g
s/"svprt\t\%\[port\]\\n\\t"/"in             r16,\%\[port\]\\n\\t"\nANDI_HWS/g
s/"o1bs\t\%\[port\]\\n\\t"/BLD_HWS\n        "out \%\[port\],r16\\n"/g
s/"o1bs\t\%\[port\]\\n"/BLD_HWS\n        "out \%\[port\],r16\\n"/g
```

and this command:

`cat ../video_gen.cpp_BACKUP | sed -f fixit.sed > /root/Arduino/libraries/TVout/video_gen.cpp`

now it works well with an 'stable' image on the monitor
i suposse it will be a bug, but i dont know where...

now next step is how to pause and resume the tv signal ... ;)  will see.
regards.-

BoRKeLSNoRKeL, 2016-12-03:
I have the same error , i''m kind of new in programming .
in one cees the code compiles and for example if i change an print command to println , the code would not compile. 
akua73 used a work around , but i dont realy know how to do that. .... just copy paste in video_gen.cpp ?
what is a sed file ?

b.t.w. i''m building a wireless timer to use on horse riding and save the time data to sd card. i alsow use a rtc to keep up with global time. to use in the time files that are being created on the sd card.
and i''m dutch that explains the bad text.

BoRKeLSNoRKeL, 2016-12-09:
Did it 👍  
copyed and replaced all the macro code in the video_gen.cpp library. 
 now it works normal in arduino 1.6.13

akua73, 2016-12-10:
Perfect 🙂👍

starling13, 2017-03-03:
I'v found that this problem have appeared after arduino IDE start to use "-flto" GCC compile and link flag.
Actually, I use not the ARduino IDE itself, but Netbeans with plugged toolchain and libs from installed Arduino IDE on Debian. After removing -flto switch from compile and link kstages I can compile for MEGA2560, using 1.8.1 toolset. From the other side, -flto flag is important and allows to forget problems in optimising and inlining the code, placed in different object modules, and ommit this option is not desirable.

toxuin, 2017-04-09:
Why this issue closed?? I just checked on latest Arduino IDE (1.8.2) and it does not compile.
Since the source comes with library bundled, why not apply this macro so it works for the latest IDE? Sed behaves differently on Mac so I can not even fix it myself.

teozfrank, 2017-06-05:
I can confirm I get the same error when using arduino 1.8.3, downgraded to 1.6.8 and it will compile.

BoRKeLSNoRKeL, 2017-06-05:
My tvout is working i can upload but would be first for me...  first sleep,
speek to you soon.

gsimpson2g, 2017-06-09:
I am running into this same issue. Going back to 1.6.8 didn't seem to fix it for me either.

khomouda, 2017-06-12:
I had the same compiling to arduino pro mini with IDE 1.8.3, and got the same asm errors, then I downgraded to 1.6.4, and it compiled well and uploaded to the board.
then, the board broke and changed it to pro mini that had lying around, but got the same errors again on the same IDE 1.6.4
can somebody that had a solution help??

Armagedon13, 2017-07-13:
Can someone pass me the cpp file?

NikitaGerasin, 2017-10-30:
@BoRKeLSNoRKeL How did you fix the error in version 1.6.13? I tried to return to 1.6.8, but this compilation error did not disappear. Also I installed 1.6.4, on which the sketch is compiled without errors, but the image is not output on TV :(
Please show the library "video_gen.cpp". Thank you.

BoRKeLSNoRKeL, 2017-10-30:
/*
 Copyright (c) 2010 Myles Metzer

 Permission is hereby granted, free of charge, to any person
 obtaining a copy of this software and associated documentation
 files (the "Software"), to deal in the Software without
 restriction, including without limitation the rights to use,
 copy, modify, merge, publish, distribute, sublicense, and/or sell
 copies of the Software, and to permit persons to whom the
 Software is furnished to do so, subject to the following
 conditions:

 The above copyright notice and this permission notice shall be
 included in all copies or substantial portions of the Software.

 THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 OTHER DEALINGS IN THE SOFTWARE.
*/

/* A note about how Color is defined for this version of TVout
 *
 * Where ever choosing a color is mentioned the following are true:
 * 	BLACK	=0
 *	WHITE	=1
 *	INVERT	=2
 *	All others will be ignored.
*/

#include "TVout.h"

/* Call this to start video output with the default resolution.
 * 
 * Arguments:
 *	mode:
 *		The video standard to follow:
 *		PAL		=1	=_PAL
 *		NTSC	=0	=_NTSC
 *
 * Returns:
 *	0 if no error.
 *	4 if there is not enough memory.
 */
char TVout::begin(uint8_t mode) {
		
	return begin(mode,255,220);
} // end of begin

/* call this to start video output with a specified resolution.
 *
 * Arguments:
 *	mode:
 *		The video standard to follow:
 *		PAL		=1	=_PAL
 *		NTSC	=0	=_NTSC
 *	x:
 *		Horizonal resolution must be divisable by 8.
 *	y:
 *		Vertical resolution.
 *
 *	Returns:
 *		0 if no error.
 *		1 if x is not divisable by 8.
 *		2 if y is to large (NTSC only cannot fill PAL vertical resolution by 8bit limit)
 *		4 if there is not enough memory for the frame buffer.
[... 755 lines cut ...]
					prescalarbits = TIMER == 0 ? 0b101 : 0b111;
				}
			}
        }
    }
    TCCR2B = prescalarbits;

	if (duration_ms > 0)
		remainingToneVsyncs = duration_ms*60/1000; //60 here represents the framerate
	else
		remainingToneVsyncs = -1;
 
    // Set the OCR for the given timer,
    OCR2A = ocr;
    //set it to toggle the pin by itself
    TCCR2A &= ~(_BV(COM2A1)); //set COM2A1 to 0
    TCCR2A |= _BV(COM2A0);
} // end of tone

/* Stops tone generation
 */
void TVout::noTone() {
	TCCR2B = 0;
	PORT_SND &= ~(_BV(SND_PIN)); //set pin 11 to 0
} // end of noTone

BoRKeLSNoRKeL, 2017-10-30:
the above is the TVout.cpp

here is the video_gen.cpp

/*
 Copyright (c) 2010 Myles Metzer

 Permission is hereby granted, free of charge, to any person
 obtaining a copy of this software and associated documentation
 files (the "Software"), to deal in the Software without
 restriction, including without limitation the rights to use,
 copy, modify, merge, publish, distribute, sublicense, and/or sell
 copies of the Software, and to permit persons to whom the
 Software is furnished to do so, subject to the following
 conditions:

 The above copyright notice and this permission notice shall be
 included in all copies or substantial portions of the Software.

 THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 OTHER DEALINGS IN THE SOFTWARE.
*/

#include <avr/interrupt.h>
#include <avr/io.h>

#include "video_gen.h"
#include "video_properties.h"
#include "asm_macros.h"
#include "hardware_setup.h"

//#define REMOVE6C
//#define REMOVE5C
//#define REMOVE4C
//#define REMOVE3C

int renderLine;
TVout_vid display;
void (*render_line)();			//remove me
void (*line_handler)();			//remove me
void (*hbi_hook)() = &empty;
void (*vbi_hook)() = &empty;

// sound properties
volatile long remainingToneVsyncs;

void empty() {}

void render_setup(uint8_t mode, uint8_t x, uint8_t y, uint8_t *scrnptr) {

	display.screen = scrnptr;
	display.hres = x;
	display.vres = y;
	display.frames = 0;
	
	if (mode)
		display.vscale_const = _PAL_LINE_DISPLAY/display.vres - 1;
	else
		display.vscale_const = _NTSC_LINE_DISPLAY/display.vres - 1;
	display.vscale = display.vscale_const;
	
	//selects the widest render method that fits in 46us
	//as of 9/16/10 rendermode 3 will not work for resolutions lower than
	//192(display.hres lower than 24)
[... 380 lines cut ...]
		"byteshift\n\t"	//18
		"byteshift\n\t"	//19
		"byteshift\n\t"	//20
		"byteshift\n\t"	//21
		"byteshift\n\t"	//22
		"byteshift\n\t"	//23
		"byteshift\n\t"	//24
		"byteshift\n\t"	//25
		"byteshift\n\t"	//26
		"byteshift\n\t"	//27
		"byteshift\n\t"	//28
		"byteshift\n\t"	//29
		"byteshift\n\t"	//30
		
		"nop\n"  "nop\n"
		"cbi	%[port],7\n\t"
		:
		: [port] "i" (_SFR_IO_ADDR(PORT_VID)),
		"x" (display.screen),
		"y" (renderLine),
		[hres] "d" (display.hres)
		: "r16" // try to remove this clobber later...
	);
	#endif
}

NikitaGerasin, 2017-10-30:
@BoRKeLSNoRKeL Thank you very much again. I fixed the libraries "TVout.cpp" and "video_gen.cpp", now the compiler produces errors about unknown names of functions and structures. Apparently the versions of our libraries do not match. Could you, in your spare time, send .rar or .zip with your library TVout? :-)

BoRKeLSNoRKeL, 2017-11-03:
I upload to my github (first time)  only my ino file is missing with
'includes' i look into that.  Look at borkelsnorkel github please.

Greetings.