sheaivey/rx5808-pro-diversity #95: rx5808-pro-osd_diversity?

Bulynia, 2016-11-06:
hi
i try to search something like rx5808-pro-osd_diversity for FATSHARK upgrade project...
but cant find nothing :(
any helps?
or how to do diversity for [rx5808_pro_osd](https://github.com/markohoepken/rx5808_pro_osd)

kabturek, 2016-11-19:
i dont know of such a project but you could add an osd screen but a lot of coding is needed - you need to connect a max chip to the arduino and to the video out and do some coding. it would probably easier to add diversity to x5808_pro_osd (add a second module and switch them)

sheaivey, 2016-12-09:
I think you are refering to Marko Hoepken's OSD project which is not diversity. You will have to make those changes yourself.

https://github.com/markohoepken/rx5808_pro_osd

closed 2016-12-09 by sheaivey