sheaivey/rx5808-pro-diversity #96: Recovery after failed flash

robert-hoffmann, 2016-11-10:
Sorry this is probably not the right place to ask ..but

I bought the built verion of this, and had a power failure while uploading the hex firmlware file, and now the board will not boot, and i can seem to reflash it either.

Is there a jumper or method to recover and be able to reflesh the firmware, or am i good to buy another one ?

Thanks for your time !
Robert

kabturek, 2016-11-19:
it would be better to ask on the rcgroups thread but if you cant flash the atmega/arduino you can check:
http://arduino.stackexchange.com/questions/13292/have-i-bricked-my-arduino-uno-problems-with-uploading-to-board
The 'built' version is similar to an arduino (with atmega328). but is programmed with usbasp.

generally its hard to 'brick' an arduino using a usbasp. try to debug if it isnt a another problem (voltage regulator etc)

sheaivey, 2016-12-09:
Questions like this should be asked over in the RCGroups thread for quicker response.

Thanks!

https://www.rcgroups.com/forums/showthread.php?2477752-DIY-Diversity-Receiver-rx5808-pro-diversity/

closed 2016-12-09 by sheaivey