sheaivey/rx5808-pro-diversity #99: 72 channels

carebear482, 2016-11-20:
I saw a new micro camera with vtx is coming out featuring 72 channels, can this be supported?
link to product: http://tinyurl.com/zaq2s8v

kabturek, 2016-11-20:
its shouldn't be a big problem to support - i could write it but i dont have any transmitter that supports it.

vadimus-ca, 2016-12-08:
The L band already in the code seems to be different from the manual that came with 72ch AIO camera/VTX.

[image]
Instead of 
5362, 5399, 5436, 5473, 5510, 5547, 5584, 5621 
it should be 
5333, 5373, 5413, 5453, 5493, 5533, 5573, 5613

Is there some calculator available to find out what corrected SPI values to put in channelTable[]?

kabturek, 2016-12-09:
it depends where you check ;-) 
https://www.rcgroups.com/forums/showthread.php?2592505-New-L-band-on-5-8Ghz

here are some tips about spi values: 
https://github.com/sheaivey/rx5808-pro-diversity/issues/75

vadimus-ca, 2016-12-09:
Actually I already calculated those values, flashed and tested L band transmission, but thanks anyway.

[image]

sheaivey, 2016-12-09:
Please pay attention to your local laws when messing with these "new bands" as they are not legal in the US. No sane US based company will sell you these VTX as the fines are quite large.

This has been discussed before...
https://github.com/sheaivey/rx5808-pro-diversity/pull/86#issuecomment-246065178

If you are going to implement these new frequencies you need to wrap the changes in a feature toggle before I will approve any PR.

Thanks

vadimus-ca, 2016-12-09:
Sure thing applicable regulations should be followed.

tipouic, 2016-12-21:
If you want the 72 chnanels, you can find the code here [https://github.com/tipouic/rx5808-pro-diversity](url)

tigert, 2017-04-05:
There are weather radars on those channels.

https://www.sciencenews.org/sites/default/files/main/articles/102215_ts_radar_free.jpg < if you start to see stuff like this on weather radar, it's not very hard to predict how nice freedoms we will have in our hobby in the future.

Those channels are illegal, also outside the US and should not be used.

https://www.youtube.com/watch?v=YipWLgu_NjY

sheaivey, 2017-05-05:
Closing this issue as there is another fork supporting those 72 channels.

closed 2017-05-05 by sheaivey