sheaivey/rx5808-pro-diversity #109: How to update your firmware

c4talyst, 2016-12-13:
I'd love to help out writing some documentation on 'How-To' guides for users to update their firmware on their devices. If anyone can give me a basic overview on what hardware & software is required I can very happily submit a PR to update the readme :)

Knifa, 2016-12-13:
That sounds super great! It's hard to be specific without knowledge of particular hardware choices, etc, but I guess on a basic level it kind of goes like the following.

- Install the latest Arduino IDE.
- Figure out which hardware you have (OLED? diversity? Nano-based? etc.)
- Update `settings.h` to match your configuration.
- Install libraries that are required for your configuration (Adafruit_GFX, TVOut, etc)
- Use an FTDI adapter (or another Arduino) to connect to the FTDI pins on your device.
- Upload the firmware.

I think maybe some walkthroughs for specific hardware would be good.

vadimus-ca, 2016-12-15:
If USBasp adapter is being used for flashing under Windows 10 libUSB**k** driver should be selected for proper firmware uploading (Sketch - Upload using programmer)
 
[image]

c4talyst, 2016-12-29:
Thanks for the tips! Running through the steps now, and have run into a stumbling block. Perhaps just a newbie error?

OS: MacOS Sierra (10.12.2)
Arduino IDE 1.8.0
Installed libraries
* Adafruit SSD1305 by Adafruit (used 1.1.2)
* Adafruit GFX Library by Adafruit (used 1.1.5)
* TVOut by mdmetzie@gmail.com (used 1.0.5)

Is this perhaps something I have neglected to configure in settings.h ?

Error:
```
Arduino: 1.8.0 (Mac OS X), Board: "Arduino Nano, ATmega328"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dan/Documents/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10800 -build-path /var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/dan/dev/github/sheaivey/rx5808-pro-diversity/src/rx5808-pro-diversity/rx5808-pro-diversity.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dan/Documents/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10800 -build-path /var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/dan/dev/github/sheaivey/rx5808-pro-diversity/src/rx5808-pro-diversity/rx5808-pro-diversity.ino
Using board 'nano' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Build options changed, rebuilding all
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "/var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409/sketch/rx5808-pro-diversity.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM/src" "/var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409/sketch/rx5808-pro-diversity.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM/src" "-I/Users/dan/Documents/Arduino/libraries/Adafruit_SSD1306" "/var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409/sketch/rx5808-pro-diversity.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM/src" "-I/Users/dan/Documents/Arduino/libraries/Adafruit_SSD1306" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src" "/var/folders/h0/39cpp_bd0vq42dw980yb47_80000gp/T/arduino_build_965409/sketch/rx5808-pro [... cut ...]
Using library Adafruit_SSD1306 at version 1.1.2 in folder: /Users/dan/Documents/Arduino/libraries/Adafruit_SSD1306 
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Using library Adafruit_GFX_Library at version 1.1.5 in folder: /Users/dan/Documents/Arduino/libraries/Adafruit_GFX_Library 
Using library Wire at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire 
exit status 1
#error ("Screen size incorrect, please fix Adafruit_SSD1306.h!");
```

c4talyst, 2016-12-29:
Excuse me, I read the error message now... Basically:

Adjust Adafruit_SSD1306.h file under the `SSD1306 Displays` heading to look like the following:
```
#define SSD1306_128_64
// #define SSD1306_128_32 // this is uncommented by default
// #define SSD1306_96_16
```

Knifa, 2016-12-29:
Glad you got it sorted! I really dislike this design choice on Adafruit's part. Needing to edit library code is super nasty.

TimothyGold, 2017-05-24:
I was able to build this update for upload but I accidentally loaded the bootloader version and I cannot update in any way now. It is using the loaded version but when I try to upload again it states that  "stk500 get sync fails"  Can someone provide a recommended solution?

Knifa, 2017-05-24:
@TimothyGold: Ouch, that sucks. I ran into the same issue with a friends' module who did the same.

I fixed it using the ISP header and [ArduinoISP](https://www.arduino.cc/en/Tutorial/ArduinoISP). The ArduinoISP might help you out (where the target is your receiver, rather than another Arduino).

What type of receiver do you have?

TimothyGold, 2017-05-24:
I have a Realacc - rx5808-pro-diversity.  Its been a while but I have a USB ASP and i think I can flash on ISP.  Once I flash the non bootloader version will I be able to use XLoader to flash updates via the FTDI interface?  Or does this require a bootloader to flash via the serial interface?

Knifa, 2017-05-24:
If you undo the screws and disconnect the top PCB with the screen, there are ISP headers underneath. 

Use your USB ASP and the ISP header to burn the Arduino bootloader. Once you've flashed the bootloader, use the FTDI pins to flash the non-bootloader version. You can continue to use the FTDI header for updates afterwards.

I guess you could also just flash the bootloader version using your USB ASP (then the FTDI will still work afterwards.)

Just, for whatever reason, trying to flash the version with the bootloader over FTDI causes weird problems.

Hope that makes sense!

TimothyGold, 2017-05-24:
This makes sense.  Thank you for your help.

TimothyGold, 2017-05-24:
One more question.  What Arduino Bootloader works best.  I see some people have tried UNO and others something else.  Any recommendations?  Thanks again.

Knifa, 2017-05-24:
Not sure on that, I used whatever it gives you with the Nano device type.

TimothyGold, 2017-05-24:
OK.  That's what I will use.

brkdwn, 2017-05-30:
I have the same issue. On both a Realacc rx 5808 split (12/2016) and a non split version (05/2017). This is where I am so far:

FTDI:
-I can't flash hex files with Xloader, it says: Upload failed
-I can compile, write, but fail to  verify with arduino

USBasp:
-I can compile and flash on ISP ports
-I can burn the bootloader (not sure if I should use nano or mega 328p, I tried both)

After burning the bootloader, I have no change with FTDI, still not possible to flash. I can only flash with xloader or arduino if I flash the same firmware witch is already on it.

So it's look like I can't manage to repair the bootloader correctly, but at least i can still flash threw USBasp. But only if I have an .ino. Is there a way to flash a .hex with USBasp ?

THX for the good work.

TimothyGold, 2017-05-30:
I was able to burn the Nano bootloader via USBasp and the ISP headers.  You can flash the .hex via the USBasp via the ISP headers.  I can now upload via FTDI fine without issue once the bootloader was repaired.  Not sure what you are facing at this point.

brkdwn, 2017-05-30:
Thank for confirming that I should be succesful. I will try further.

Can you tell me how are you flashing hex file with USBasp ?

TimothyGold, 2017-05-30:
Its been a while but I think you can pick the USBasp as the programmer in the Arduino IDE and then program like you normally would via the ISP header.  You will have to do all programming via ISP header.

brkdwn, 2017-05-31:
Ok It was a driver issue, now the bootloader is working again, an i can flash witfh ftdi again

thx

pbootsma, 2017-08-28:
I just want to add that I finally got the hex flashing with USBASP working on MacOS 10.12 Sierra with the following command if Arduino IDE (1.8.4) is installed:

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cusbasp -Uflash:w:Downloads/rx5808-pro-diversity-v2.0-with-bootloader.hex

You have to omit the -D parameter (Disable auto erase for flash memory), otherwise you get verification errors.