sheaivey/rx5808-pro-diversity #112: Break out SPI commands into own file

sheaivey, 2016-12-20:
Simplified receiver SPI command function.

Knifa, 2016-12-20:
So I've got this one already in this [commit](https://github.com/sheaivey/rx5808-pro-diversity/pull/106/commits/e96217aee8795f5c45be55dd6c0fc0b555926d64) but I'm keen to move those out further.

sheaivey, 2016-12-21:
If i remember correctly the current code is just bit-banging out the commands and does not even need the SPI library. That said the bit-banging code is very dirty haha.

Knifa, 2016-12-21:
Yep, I was wondering if I could maybe simplify by just using whatever Arduino has already but it turns out bit banging is totally spot on. I tidied it up a bit at least. :)

Knifa, 2016-12-23:
https://github.com/sheaivey/rx5808-pro-diversity/pull/106/commits/1dbce46f2668854827628454ae9300d1de69cf8e

Moved these out a bit more now --- not sure if calling the broken out files "driver" is the best bet since the other file still deals with RSSI reads etc. Bit nicer though!

Knifa, 2017-03-13:
Closing this as I think they've been moved!

closed 2017-03-13 by Knifa