sheaivey/rx5808-pro-diversity #113: Refactor main loop into smaller functions and files so it's easy to follow.

sheaivey, 2016-12-20:


Knifa, 2016-12-20:
Some general unsorted thoughts around this:

- Ideally split up so states are non-blocking (i.e., no do-while loops for locking in a particular state). Means we can still arrange for other stuff to happen easily (like update RSSI values in one place, do diversity switching in one place, etc.)
- Need a way to deal with button presses centrally given the above. I'm thinking maybe heading towards somehow event based? Not sure how the overhead would work out on this on the AVR chips.

I will try head for literally splitting it up as is for now, though.

sheaivey, 2016-12-21:
Those are some good goals and are some of the first things I did when creating the La Forge FPV fork as the main loop was a disaster.

Knifa, 2016-12-23:
Starting to make a go at this now. Keen to hear thoughts on the way I'm doing the state machine.

https://github.com/sheaivey/rx5808-pro-diversity/pull/106/commits/d997e5960d3552a5b208843b3094249bd4bcb920

I wanted to avoid using huge switch statements where possible but I'm aware of the potential issues with memory usage. Still trying to work it out as I'm going along!

Knifa, 2016-12-23:
https://github.com/sheaivey/rx5808-pro-diversity/pull/106/files#diff-dc7eb18d61c1335e0c1a37dba8755681 good example here too, maybe?

Knifa, 2016-12-23:
Making good progress on this. Manual, seek, bandscanner and screensaver all working! It's made it so much faster to get stuff working at the very least.

closed 2017-03-13 by Knifa