sheaivey/rx5808-pro-diversity #114: Refactor GUI and menu logic.

Knifa, 2016-12-20:
That was what got me started in the first place since I found it so hard to make new menus! Looking to  refactor GUI code and menu logic such that new menus can be added and modified easily, along with generally tidying stuff up.

General ideas:
- Make a mini-toolkit for making menus or arbitrary size, length, and control (e.g., support character field or numerical values elsewhere).
- Possibly use hooks for the above so you can attach functions to particular menu items (rather than detecting specific menu IDs and doing stuff based on that in one big function)
- Make it easy to deal with overall.
- (Possibly a big one, for another issue perhaps) Generalise enough that OLED and TVOUT can share the same/similiar code.

I have a bit of an issue here where my device doesn't have TVOUT by default so I might need to bodge it a little bit so I can test and refactor this too.

sheaivey, 2016-12-21:
Breaking down the views into basic components will help tremendously and help standardize the basic look and feel. If you can make these components responsive based off screen width/height you will have a big step towards being able to reuse the screens for different screen resolutions (TVOUT/OLED).

You will also run into the fact that the TVOUT library does not follow the same graphics style as the Adafruit graphics library. Honestly it would be easier to re-create the TVOUT library to extend the Adafruit graphics library like the OLED library does. All that would need to be created would function how a pixel gets drawn onto the TV screen and Adafruit handles the draw functions and screen buffer. (This would require some more in-depth knowledge on how to generate NTSC/PAL signals)

Then you just swap which library is being used in the settings.h

Those are just some more ideas I had to help bridge the gap between OLED/TVOUT but by no means something you have to take on.

-Shea

sheaivey, 2017-05-15:
This has been accomplished with the release of v2.0

closed 2017-05-15 by sheaivey

balrog-kun, 2017-06-08:
To be clear, the part about TVout support isn't done, right?  As far as I can tell there's no support code anymore.

Is anyone working on re-adding that?