sheaivey/rx5808-pro-diversity #116: 6 way diversety

chilly24, 2017-01-03:
Hi,
is it possible to build a six way diversity like the Quanum Overlord?
https://www.youtube.com/watch?v=EKccpoHRunc

I mean i just need to connect 6 moudles to a ardunio? of course the firmware needs to be reprogrammed

Cheers

Knifa, 2017-01-03:
At the moment the software doesn't support diversity setups beyond two receivers at all and it's pretty ingrained. A good bit of work would be needed to make it work (but might be OK after the tidy up).

This is a dupe of #108 which has some information about a fork that may suit your needs.

Knifa, 2017-01-03:
To add to the above, I would totally be up for making it generic but I don't have the hardware to test it. Need an AVR chip with more pins and a switch IC with more channels.

closed 2017-01-03 by Knifa

VictorSchonk, 2017-01-10:
Would it be possible to write code to make it scalable if the microcontroller (processor) would allow it with the pins and processing power?