sheaivey/rx5808-pro-diversity #118: help

boothbenny, 2017-02-27:
help total new to this and having problems with getting the code onto my nano 

if this help

Arduino: 1.8.1 (Windows 7), Board: "Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bob\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10801 -build-path C:\Users\bob\AppData\Local\Temp\arduino_build_384050 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\bob\Documents\Arduino\sketch_feb27a\sketch_feb27a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bob\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10801 -build-path C:\Users\bob\AppData\Local\Temp\arduino_build_384050 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\bob\Documents\Arduino\sketch_feb27a\sketch_feb27a.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_GFX_Library" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\bob\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\sketch\sketch_feb27a.ino.cpp" -o "C:\Users\bob\AppData\Local\Temp\arduino_build_384050\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\bob\Documents\Arduino\sketch_feb27a\sketch_feb27a.ino:58:21: fatal error: screens.h: No such file or directory

 #include "screens.h"

                     ^

compilation terminated.

Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
Using library Adafruit_SSD1306 at version 1.1.2 in folder: C:\Users\bob\Documents\Arduino\libraries\Adafruit_SSD1306 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.1.5 in folder: C:\Users\bob\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
exit status 1
Error compiling for board Arduino Nano.

Knifa, 2017-03-05:
Did you clone the repository or are you trying something else out here? (Copy and paste?) Sketch filename doesn't match the one in the repository and you're missing other files.

https://github.com/sheaivey/rx5808-pro-diversity/tree/master/src/rx5808-pro-diversity

You need all of these.

Knifa, 2017-03-13:
Going to close this since there's been no reply. Feel free to comment if you have anything else to add @boothbenny!

closed 2017-03-13 by Knifa