sheaivey/rx5808-pro-diversity #126: Re-implement beeper feedback

Knifa, 2017-05-15:
The beeper doesn't beep any more when you press buttons or do things.

I think the behaviour before was:
- Beep on any button press.
- Beep on "system" activity like saving or low voltage.
- Beep on low RSSI.

We can make use of the button hooks to beep on button press, but perhaps it should only beep when it actually does something?

sheaivey, 2017-05-15:
On the LaForge we beep when rx modules switch and button presses.

lovejoy777, 2017-05-21:
beep on rx switching,  low rssi and voltage alarms would be good. personally not bothered about button press (although would be nice) as my diy hardware has the 3 tactile switches which click quite loud when pressed. Just my 2 cents 😉. keep up the great work guy's.