sheaivey/rx5808-pro-diversity #129: Update with module installed

19Gerhard85, 2017-05-15:
Hey guys,

I have quick question: Is it ok, to update the module while it's installed in the goggles (not powerd up)?

Greets

Knifa, 2017-05-15:
It depends on your particular module but I would recommend taking it out. The RealAcc version tries to backpower the goggles through the 5V rail (without success!).

19Gerhard85, 2017-05-15:
I'm using the Realacc module. Can the backpower damage the googles?

Knifa, 2017-05-15:
It could make the voltage regulator a little upset, but not sure about real damage. The main problem is that (assuming it's powered over USB) your power source might go overcurrent.

Still strongly advise just taking it out!

19Gerhard85, 2017-05-15:
Ok. Thank you very much!

closed 2017-05-15 by 19Gerhard85