sheaivey/rx5808-pro-diversity #136: No video output

teozfrank, 2017-06-05:
Hi,

I am using the RX5808-pro-basic and I cannot get video to display, I have it wired as shown and have the sketch on my arduino mini. I am using a PAL monitor, could you offer any advice?

Kind Regards,
Frank

sheaivey, 2017-06-05:
You will need to use Release v1.2 for video instead if you are not using the OLED display.

https://github.com/sheaivey/rx5808-pro-diversity/releases/tag/v1.2

teozfrank, 2017-06-05:
Ah perfect thank you!

Only problem is, I get a blank screen now instead of no signal, any ideas?

[image]

teozfrank, 2017-06-05:
Ok I got it, just needed to press a button, awesome thank you!

closed 2017-06-05 by teozfrank