sheaivey/rx5808-pro-diversity #138: SH1106 compile error

guelz, 2017-06-23:
trying to compile with SH1106 OLED enabled Ikeep getting this error:

> Arduino: 1.8.3 (Windows Store 1.8.6.0) (Windows 10), Board: "Arduino Nano, ATmega328"
> ----------------snip--------------------
> settings_internal.h:56: error: 'Adafruit_SH1106' does not name a type
> 
>    #define OLED_CLASS Adafruit_SH1106
> 
> C:\Users\pguel\AppData\Local\Temp\arduino_build_105665\sketch\ui.h:23:12: note: in expansion of macro 'OLED_CLASS'
> 
>      extern OLED_CLASS display;
> 
> Using library Adafruit_SH1106-1.1.0 at version 1.1.0 in folder: C:\Users\guelz\Documents\Arduino\libraries\Adafruit_SH1106-1.1.0 
> Using library SPI at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI 
> Using library Adafruit-GFX-Library-master at version 1.2.2 in folder: C:\Users\pguel\Documents\Arduino\libraries\Adafruit-GFX-Library-master 
> Using library EEPROM at version 2.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\libraries\EEPROM 
> Using library Wire at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\hardware\arduino\avr\libraries\Wire 
> exit status 1
> 'Adafruit_SH1106' does not name a type

Does someone know how to set that "type"? I'm still pretty noobish!

DanNixon, 2017-06-24:
Looks like you have not installed the correct libraries. See the Wiki for step by step instructions (https://github.com/sheaivey/rx5808-pro-diversity/wiki/Guide:-Compiling-From-Source).

pcipri76, 2017-07-01:
I think I have the same problem to compilation like I write here:
https://www.rcgroups.com/forums/showthread.php?2477752-DIY-Diversity-Receiver-rx5808-pro-diversity/page122

In file included from C:\Users\Darius\Desktop\rx5808-pro-diversity-develop\src\rx5808-pro-diversity\rx5808-pro-diversity.ino:38:0:

settings_internal.h:56: error: 'Adafruit_SH1106' does not name a type

   #define OLED_CLASS Adafruit_SH1106

                      ^

sketch\ui.h:23:12: note: in expansion of macro 'OLED_CLASS'

     extern OLED_CLASS display;

            ^

exit status 1
'Adafruit_SH1106' does not name a type

Nandox7, 2017-08-18:
That error is due to the fact it doesn't load the SH1106 header file.
Add the following to the ui.h file, could be just bellow the include for the SSD1306 OLED.

---
#include <Adafruit_SH1106.h>
---

Remember that you'll need to manually add the SH1106 Adafruit library to your arduino instalation.

Nandox7, 2017-08-18:
#141