sheaivey/rx5808-pro-diversity #139: RX5808 V07 20160524 SPI Mod needed?

stray77, 2017-07-10:
I opened the can on one of my new  RX5808 V07 20160524 modules to remove the resistor to enable SPI only to find there was no resistor to remove.
I'd just like to confirm this module needs no modification before I put the lid back on the can.
Here's pics of the back and under the can.
[Back of RX5808 V07](https://image.ibb.co/hmpJkF/Snapshot_20170709_4.jpg) and [Inside the can](https://image.ibb.co/fP6W5F/Snapshot_20170709_3.jpg)

mausohr, 2017-07-11:
On Sun, 2017-07-09 at 17:08 -0700, stray77 wrote:
> I opened the can on one of my new RX5808 V07 20160524 modules to
> remove the resistor to enable SPI only to find there was no resistor
> to remove.
> I'd just like to confirm this module needs no [...]

I ordered similar modules from banggood:
https://www.banggood.com/FPV-5_8G-Wireless-Audio-Video-Receiving-Module-RX5808-p-84775.html. The imprint on the rear looks a bit different from your modules, but the layout seems to be the same.

The banggood modules also already miss the resistor and can be accessed
via SPI unchanged (I verified), although this fact is not mentioned in
any way on the banggood page.

stray77, 2017-07-11:
Thanks very much.  Nothing about SPI was mentioned in the ebay listing either.  I'd like to see this mentioned in the docs but I guess I could have just tried the module to see if it worked before I opened it up to do the mod.

closed 2017-07-13 by stray77