sheaivey/rx5808-pro-diversity #140: TVout,Can't compile

Armagedon13, 2017-07-13:
hi i can't compile, and yes, i read the closed issue, but, i don't have linux for remplace the archive. Can someone pass me the cpp file?

stray77, 2017-07-19:
Does this help?  I really can't tell what you're looking for.
https://github.com/Avamander/arduino-tvout

caveymon, 2017-08-16:
I had this issue too. Some compiler switches have changed between arduino ide versions, and newer (1.8+) don't seem to support macros anymore. 

I patched the tvout library included in this project on my own local copy. Haven't bothered to push the changes, maybe I should somehow.

havent been able to test it yet, but I needed it for a small modification I did to the schematics. Not sure if it has the sorted effect though :)

GregTronics, 2017-10-03:
Per the closed thread, the way to fix this error is to go into the video_gen.cpp file and replace all the asm macros with the asm functions.  My patched one is attached.  Works with newest compiler and can confirm the screen works again.    
[video_gen.zip](https://github.com/sheaivey/rx5808-pro-diversity/files/1351145/video_gen.zip)

johnbentcope, 2018-04-14:
GregTronics, replacing video_gen.cpp in the library folder results in new errors, mostly stating "Error: macro 'macroNameHere' was already defined. Any thoughts?