sheaivey/rx5808-pro-diversity #153: How are the RSSI Values created?

swatza, 2018-07-10:
I was trying to dig through the code as I am using this as a low cost sensor for some RF research to figure out what the outputted value represents. The best guess I have is that its an RCPI 8bit between 0-220 (as comments I found indicated min values were 90 and max were 220) but I am not sure if that mapping is correct. 

Any information about the exact details would be helpful.

JyeSmith, 2018-07-10:
In receiver.cpp you can see the analogRead function which reads the RSSI value.

The below link explains the returned value 😃

https://www.arduino.cc/en/Tutorial/AnalogInput

swatza, 2018-07-10:
Okay, so if I am understanding this correctly: mapping 0-1023 from 90 to 220 or 0 to 100. Therefore 90 would equal 0 Volts and 220 should equal 5V or 100%.

JyeSmith, 2018-07-11:
The RSSI min/max output range isn't from 0 to 5v, but more like 0.5 to 1.2.  Hence the min and max calibration values are only ~90 to ~220.

The 0 to 100 conversion you see in the code is to normalise Rx modules on the same scale. It then makes it easy to see a 2% difference, which is the threshold for switching between the Rx modules.