sheaivey/rx5808-pro-diversity #156: SPI OLED Screen?

akosfonod, 2018-12-16:
I have ordered by mistake an SPI version of an OLED screen (SSD1306), beside the fact that the module itself can be re-configured with some additional small SMD resistors,but it was pretty easy to surround the display constructor with some defines and call the appropriate SPI version constructor of the adafruit driver lib. I have already tested the SPI version it works flawlessly.
Is anyone intrested? Is this project still alive?

JyeSmith, 2018-12-17:
SPI should be much faster than I2C, particularly when its not bit banged.  So it would be useful to speed up the loop.  But everyone has moved onto STM32F1 which is much faster.

With that said its pretty dead in here :(

Personally I dont see the need for the OLED and have removed it from my VRx.

akosfonod, 2018-12-17:
The way how I have modified it is using the HW SPI, so no bitbanging.
@JyeSmith Can you pls name the  STMF1 repo?
Thank you :) 
I will close the the Issue afterwards.

JyeSmith, 2018-12-17:
piodabro/WPro58 has an active community. They concentrate on the Eachine module and adding OSD functionality.

JyeSmith/FENIX-rx5808-pro-diversity is my fork.  The recent master is for quadversity, but at the moment Im concentrating on using a touchpad and improving in goggle menu functionality (check the touch (STM32F1 based) and esp32 (**this is more active**) branches).  Hence the comment about the OLED being redundant.

There are a couple more which havent been active in over a year.

closed 2018-12-17 by akosfonod