sheaivey/rx5808-pro-diversity #158: help wanted: how can it be changed to communicate with another module in quadversity?

laurentopia, 2019-04-23:
I'm thinking ev200d here, their modules are based on this project it seems. They suck so I'm wondering if this can be changed in firmware so both modules communicate and decide which has the best RSSI and switch globally to it (in the ev200d).
Any help appreciated.

JyeSmith, 2019-04-23:
https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/tree/quadversity

This is a fork I developed a little back adding quadversity to the rx5808-pro-diversity.  I have no idea how the EV200 is set up to do it, but if you spend some time with a multimeter poking around you could figure it out and report back :D

laurentopia, 2019-04-23:
Thank @JyeSmith.
I've never done that sort of poking but the goggles are open so it's a good chance to do it, what do you recommend I start with to see what's needed?

JyeSmith, 2019-04-23:
It mightn't be so easy if you've never done it before and no familiar with module schematics.  But at a minimum find out what the module-to-goggle sockets connect to.

laurentopia, 2019-04-24:
Do you have a pinout of a standard fatshark module?

JyeSmith, 2019-04-24:
Check the readme for the connections.

laurentopia, 2019-04-24:
The readme contains only the connections of each components, not the pinout to the goggles

JyeSmith, 2019-04-24:
https://www.rcgroups.com/forums/showatt.php?attachmentid=11192179

laurentopia, 2019-04-25:
Thanks!
Did you build your project with 4 rc5808 inside one module?
I see that it could be turned into 2 diversity modules that connects like that image, (is it the source of Lagorge?)
Since many people are getting rid of their default ev200d module (some batches loose lock mid air) I'm thinking that would be a great alternative.

JyeSmith, 2019-04-26:
I did build an Arduino based quadversity Rx and a STM32 based module for Fatsharks.

https://imgur.com/a/yCRRbOW

laurentopia, 2019-04-26:
Quite a contraption, which one am I looking at and how does it "fly" ?

JyeSmith, 2019-04-27:
This is a prototype STM32 module. Works just like a diversity with more patch antennas :)

laurentopia, 2019-04-28:
nice long coverage :) how fast is commutation?

JyeSmith, 2019-04-29:
Current switching rate is about 15kHz in 2 antenna diversity mode.

laurentopia, 2019-05-02:
So do you stitch the image per line?
I thought Xversity was commutating at the end of the last scanline.

brainbubblersbest, 2020-02-22:
Hello, figured out that the stock modules are communicating over SPI at the first Three FS module Contacts.
The Resistors on the 5808 Modules are not existent. (removed the shielding)
And the Ch1, Ch2, Ch3, Contacts from each RX are connected to the three first contacts Button 1,2,3 in module bay. With 0,1ohms!

Also the RSSI is not output to the pin 8 of FS Bay.
Seems Signal Quality will only reported over SPI?

I figured out that RSSI is connected to this tiny Board upside from each RX5808 Boards. 
[image]
[image]

Can Provide measurement with scopemeter. Have two Pro58 Boards.
Maybe some Hardware modification is needed to allow communication for STM32 to the Google.