sheaivey/rx5808-pro-diversity #159: Open pilot branded rx5808 pro diversity

stevefoxuk, 2019-06-18:
Ive tried to find info on this but cannot find anything definitive.

Can this firmware be flashed to the openpilot branded rx5808 pro diversity module?
this is the exact unit im talking about:  https://www.aliexpress.com/item/32703859969.html?spm=2114.search0604.3.48.748e227e3IHnEe&ws_ab_test=searchweb0_0%2Csearchweb201602_9_10065_10130_10068_10547_319_317_10548_10696_10192_10190_453_10084_454_10083_10618_10307_10820_10301_10821_10303_537_536_10059_10884_10887_321_322_10103%2Csearchweb201603_53%2CppcSwitch_0&algo_expid=5bf5eed3-0a8f-4fc7-be3b-7b895aba171b-6&algo_pvid=5bf5eed3-0a8f-4fc7-be3b-7b895aba171b

Thanks in advance, steve.

JyeSmith, 2019-06-18:
I would be very suprise if it couldn't.

stevefoxuk, 2019-06-30:
[image]

This is the board i would like to know if i can flash this firmware too, i have examined the board against Shea Ivey designed boards and they are 'almost' identical play the component values are the same as are the two IC's on the board but i want to make sure and not risk bricking the board!

mozgoverflow, 2019-10-23:
Hi,
I have similar module.

[image]

Flashed with latest release (built from sources in Arduino IDE 1.8.9) and flashed via USBasp programmer. 

At first look, it works, but I have strange bug on OLED: image is interlaced and stretched vertically 2x.

What things I could try to fix this?

mozgoverflow, 2019-10-23:
Update:

I have flashed prebuilt HEX (from Releases) and now it works without OLED issues.

stevefoxuk, 2019-10-23:
We both have the exact same unit, mine only looks slightly different because I have removed the oled and soldered in pins so the screen can be remotely mounted and I've added those heatsinks myself for cooling :)

I finally managed to get mine running with this firmware,  what I'd really like to do is get it running with a 1106 oled instead of the 1306 the firmware natively supports.

I want to use a 1.3" oled but can only find them in 1106 controller types :(

arvebjoe, 2019-12-01:
How did you connect this up? What are the TX\RX pins? Or did you use the SPI port (miso, mosi and sck)?

stevefoxuk, 2019-12-01:
> How did you connect this up? What are the TX\RX pins? Or did you use the SPI port (miso, mosi and sck)?
> 
> Update:
> I found an article describing the process:
> https://hoogvlieger.wordpress.com/2017/08/13/how-to-update-ge-fpv-rx5808-firmware/amp/

If you were talking to me then yes, I soldered header pins to the spi pads and flashed it that way :)

pavel4n, 2020-03-28:
Hello, 
Could anybody help?

How do you power the device?
Is it via 5V or is it possible via 12V somehow somewhere?

Where do you take video and oudio output?

Thanks.

stevefoxuk, 2020-03-28:
> Hello,
> Could anybody help?
> 
> How do you power the device?
> Is it via 5V or is it possible via 12V somehow somewhere?
> 
> Where do you take video and oudio output?
> 
> Thanks.

If you look at my pic in a previous post, the pins for audio and video are labled on the board.

I believe these boards can be powered on the first pin on the left with higher voltages as that pin goes to a regulator but please don't hold me too that, I have only ever powered mine with 5v via the 5v pin.

xtremcoimbra, 2021-02-13:
Hi,
Can anyone help me
I have a board that´s works well with a new 1.3 "oled sh1106, but I'm not able to compile with the oled version to get these menus:

https://github.com/sheaivey/rx5808-pro-diversity/blob/develop/docs/oled-arduino-nano.md

However if using the SSD1306 it compiles perfectly in this version.

I was grateful for the possible help

xtremcoimbra, 2021-02-13:
Resolved, This is an issue with the Dim command on sh1106. Uncomment this command everything is fine