sheaivey/rx5808-pro-diversity #162: Hello

gnurc, 2019-12-31:
Hello,

I have some years working in a VTX and VRX modules, this year I have assembled a VRX with high gain.

Could it be adapted to this firmware?

I am working on another similar project and I don't have much time to work on the code, besides my idea was to do something open and collaborative.

Best regards
[image]
[image]
[image]

JyeSmith, 2019-12-31:
Sport awesome to see someone working on an rx5808 replacement!

If you are using the same rtc7615 then your receiver module will work with this project.

I've built a module based on the ESP32 which is a fork of this repo.  Happy to test out your module if you have some samples.  Jump on my repos discord :)

https://github.com/JyeSmith/FENIX-rx5808-pro-diversity

gnurc, 2020-01-01:
Hi,

Thank you,

Yes I use several rtc integrated circuits.

I have a few assembled units, the first one I did in july. I have around 30 pcb but only a few assembled. If you send me an address I could send you one to try. Now I am doing the pcb so I hope to improve the design of the pcb and I wait to try with rf copper laminates to see if that improve something.

Also I have ic to a 2.4Ghz version if that may interested.

Awesome I don't know that exist a ESP32 port. I'm going to look it. I have not try this firmware, I have a arduino micro only not sure if works, I use kinetis or lpc ic.

Best regards and Happy new year

JyeSmith, 2020-01-01:
@gnurc jump on https://discord.gg/gG2sjjK and we can direct message each other.

What makes your module have a higher gain?  Different SAW filter?

gnurc, 2020-01-15:
Hi,

Sorry, I haven't seen the last post. 

Double LNA. By the moment the SAW filter is Murata.

BR.