sheaivey/rx5808-pro-diversity #164: Start-up problems ... I have no signal on the screen

juanvivo, 2020-04-29:
I have managed to assemble the complete circuit but I can't get it to work ... Mount the components on a breadboard, load the code in the arduino nano from the ide, and when feeding the circuit I don't get anything.

The status LED lights up but goes out instantly. The receiver C LED lights up and nothing else happens. I try to press the fourth button to see if I get a video signal from the menus or some other but I am not successful. I can't see anything on the screen ...

Any ideas what I may be doing wrong?

JyeSmith, 2020-04-30:
One thing you can check is if the rx5808 modules are working.  With no SPI connection they will default to A1.  Only connect power and video, then set your VTx to A1 and check for an image.  Now check the RSSI pad for a voltage change when switching the VTx between A1 and something else.

juanvivo, 2020-04-30:
okay. I will start testing as you indicate.

Shouldn't the channel search and configuration menu be shown on the screen even if the rx5808 doesn't work?

juanvivo, 2020-05-07:
Can you tell me a version of the arduino ide that does compile the code?

juanvivo, 2020-05-13:
What should happen when it is turned on for the first time?

I get video signal video signal on the monitor, A1 signal in the tx. I see the image on the monitor but I press the buttons and nothing happens ... Pressing the osd button should the configuration menus come out?

JyeSmith, 2020-05-13:
You have followed the wiring diagram from here? https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/tree/quadversity

If so, after pressing the OSD button you should see the menu in your goggles in stead of video.

juanvivo, 2020-05-14:
Yes, I have made that diagram, and I have recorded the code here: https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/archive/quadversity.zip

I am going to try to create a pcb board on the mill to eliminate possible connection failures on the breadboard ... Are there board files available?

JyeSmith, 2020-05-14:
No PCBs for Arduino nano or stm32f1 quadversity.  I have PCBs for the ESP32 module but I did not continue quadversity with that branch.

juanvivo, 2020-05-14:
Well I can't think of what to do to find the problem apart from making a pcb ... I've never done it, but I'll try. If you have any more ideas about what could be wrong ... I will appreciate it.

JyeSmith, 2020-05-14:
So when you press the OSD button the video stays? Or disappears? Or gets worse?

If you have discord jump in the chat linked on my repo.

juanvivo, 2020-05-14:
i have discord, but i cant see any link to your chat... I am not very familiar with the github format and I have a hard time finding information ...

JyeSmith, 2020-05-14:
https://discord.gg/gG2sjjK

juanvivo, 2020-05-14:
Solved. The problem was that the buttons were upside down ... and in this way they were always pressed ...