sheaivey/rx5808-pro-diversity #165: I have errors compiling and saving the code in arduino

juanvivo, 2020-05-11:
I am having trouble saving the code to the arduino. I have tried several versions of the IDE (1.6.8, 1.6 and 1.5) but I have not been successful with any ... Can you tell me a version of the IDE with which you are sure you can compile and save the code without errors?

JyeSmith, 2020-05-11:
Looks like the last time I committed to quadversity was July 2018.  Try 1.8.5 which was the current release back then.

https://www.arduino.cc/en/Main/OldSoftwareReleases
https://www.arduino.cc/en/Main/ReleaseNotes

juanvivo, 2020-05-12:
TKS @JyeSmith 

I installed version 1.8.5. Also the libraries that are attached in the rar and the adafruit 1306 and gfx libraries that this requests. Many errors remain that there are redefinitions.

This is the first to come out, but there are many more similar ones:

Arduino:1.8.5 (Windows 8.1), Tarjeta:"Arduino Nano, ATmega328P"
In file included from sketch\ui.h:7:0,

                 from F:\FENIX-rx5808-pro-diversity-quadversity\FENIX-rx5808-pro-diversity-quadversity\src\rx5808-pro-diversity\rx5808-pro-diversity.ino:44:

C:\Users\Elena\Documents\Arduino\libraries\TVout/TVout.h:50:0: warning: "WHITE" redefined

 #define WHITE     1

 ^

crysis-ps, 2020-06-24:
hello @juanvivo 
can you please say, wich versions of libraries did you install?
seeems i tried many combinations of adafruit 1306 and gfx libraries, but still have warnings in compilation (and not working receiver).
I tried 1306 - 1.2.7 (and all near it) and GFX 1.2.9-1.4.14 - but no success

juanvivo, 2020-06-24:
all libraries are in the latest version except one. But I don't remember which one, when I get home I look at it and tell you ...

juanvivo, 2020-06-24:
the version that worked for me was adafruit 1306 july 2018. It seems to me to remember that later I tried with some later versions and it also worked well.

crysis-ps, 2020-06-24:
thank you for the answer.

I can found:
Adafruit_SSD1306-1.1.2 2016-12-05
Adafruit_SSD1306-1.2.0 2018-11-08
I cannot find the version from jule, so seems it should be 1.1.2 version...
GFX - the latest?
And what branch of code did you build? develop or master?

crysis-ps, 2020-06-24:
Tried all this versions, but still unsucsessfull (get many warnings during compiling, and then white snow on the screen). It would be fine, if you find the working versions :)
IDE 1.8.5 and 1.8.12

juanvivo, 2020-06-25:
The versión that i use without problems is 1.3.0 in 1.8.12 ide .

truglodite, 2020-12-02:
> The versión that i use without problems is 1.3.0 in 1.8.12 ide .

I was also having compiler errors until I read this. I downgraded from the current Adafruit GFX lib (1.10) to 1.3.0, and it compiles now.

crysis-ps, 2020-12-11:
I also found working versions. this was Adafruit_SSD1306 1.1.2 and Adafruit_GFX 1.2.2

JohnWatson0, 2021-04-17:
When I compiled the source code, I reported this wrong, what's going on?

new.cpp.o (symbol from plugin): In function `operator new(unsigned int)':
(.text+0x0): multiple definition of `operator new(unsigned int, void*)'
sketch\state.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1

largrovas, 2021-11-07:
> When I compiled the source code, I reported this wrong, what's going on?
>
> new.cpp.o (symbol from plugin): In function `operator new(unsigned int)': (.text+0x0): multiple definition of `operator [...]

I had the same error, but just now I've been able to compile, just downgrade your Adafruit_SSD1306 and Adafruit_GFX libraries. The versions that worked for me are Adafruit_SSD1306 1.1.2 and Adafruit_GFX 1.2.3