sheaivey/rx5808-pro-diversity #166: Help Identifying a module - Based on this project?

adamprescott, 2020-05-11:
Hi,
I'm looking for some identifying if this module is compatible with this project and I'm not sure how old it is or what branch I'd need to use, here's the a video of it https://photos.app.goo.gl/qy6soP9KuvvAzzJp8

I think it might be running an older version of https://github.com/JyeSmith/FENIX-rx5808-pro-diversity by @JyeSmith would be good to know the capabilities of the hardware before I go flashing newer code to it :)

Thanks in advanced for your help.

JyeSmith, 2020-05-11:
Follow up for any future users of this module.  **Master** is the correct branch for this VRx.

Use REALACC_RX5808_PRO_PLUS_OSD.hex from https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/releases/tag/v0.2

Install following this guide (e.g. just like a flight controller) https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/wiki/Installing

[image]

adamprescott, 2020-05-11:
Thanks for your help on your Discord @JyeSmith your firmware works a treat :D