sheaivey/rx5808-pro-diversity #167: Running the code on arduino pro mini 328p

crysis-ps, 2020-06-22:
Hello, I have a question. Could this code be working on arduino pro mini 328p 16Mhz 5V ?
Seems it has same proc - atmega 328p as on arduino nano. All required pins available.
I built the pcb board with pro mini (same as simple oled non diversity version). And trying to flash it.
But it has some problem. After flashing with arduino IDE, I get white snow screen on oled, the status led just blink once after startup. And there is no any reaction on buttons.
May be something wrong with libraries? I have latest adafruit 1306 and Gfx libraries.
(the adafruit 128-64 i2c example work fine on this board)

crysis-ps, 2020-06-24:
flash the compiled hex - and seems all working.
Now I try to find required version of libraries to successfull compiling code..

crysis-ps, 2021-02-16:
with properly versions of libraries it works fine

closed 2021-02-16 by crysis-ps

largrovas, 2021-11-07:
> with properly versions of libraries it works fine

For me: Adafruit_SSD1306 1.1.2 and Adafruit_GFX 1.2.3