sheaivey/rx5808-pro-diversity #169: Is this still being worked on?

appkoders, 2021-08-06:
Are there plans on continuing the work on this project?

If anyone knows of a fork that added the voltage alarm please post the link or if there is a better project out there like this. I started coding on adding the voltage alarm not yet done but wanted to ask here because it is a waste of time if someone already did it.

JyeSmith, 2021-08-06:
I added voltage monitoring.  But my current fork is pretty different.

https://github.com/JyeSmith/FENIX-rx5808-pro-diversity

cad435, 2021-11-14:
I just forked and builded the project. 
Switched to PlatformIO Enviroment, added code so that the beeper is beeping whenever voltage falls under 3.2V

I also added the schematic I created. It has a builtin Lithium charging circuitry with a stepup to boost the voltage to 5V.

You can find it here and start from there if interested: https://github.com/cad435/rx5808-pro-diversity

keep in mind that out of time constraints its not the tidiest project of mine tbf...